deinstapel/cpupower #8: Show current cpu freq in top bar

technixp, 2015-04-12:
It would be very nice to have checkbox for this option.

martin31821, 2015-04-12:
With this option enabled you could get serious log spam into your `auth.log`, as policykit logs every invoke of the cpufreqctl utility. I don't know how to prevent this yet, if I can figure this out, I could add a switch for this.

technixp, 2015-04-13:
What do you think about this approach?
https://github.com/paradoxxxzero/gnome-shell-system-monitor-applet/blob/master/system-monitor%40paradoxxx.zero.gmail.com/extension.js#L1348

martin31821, 2015-04-13:
Nice approach, I tested it and it seemed to work, so I will add this to the next version, therefore you won't get any spam to your `auth.log` even if the extensions menu is open. 
This will be configurable.

closed 2015-04-15 by martin31821