deinstapel/cpupower #11: It hangs my GNOME 3.16

steelbrain, 2015-05-27:
It hangs immediately after it, the TTY{X} works though. Moving the cursor works as well, nothing else does. GDM works too. Restarting doesn't help so I removed the extension from FS.

martin31821, 2015-05-30:
Can you include some logs from GNOME? (especially the output of `journalctl -xb`

stuaxo, 2015-06-16:
Just confirming I had something similar to this .... worked, but I hadn't copied in the external tools yet ... copied them in and gnome shell just hung on launch.

Can upload the journalctl -xb output somewhere, its 1.5mb ...

martin31821, 2015-07-18:
This problem should be fixed as of V2, as no more installation of cpufreqctl is required. 
If you can confirm this, please close this issue.

spacepluk, 2015-07-19:
I'm having the same problem, and I just tried the last commit in master but it still hangs.

mumblyOMOD, 2015-07-20:
I can confirm. I have the exact same symptoms as the original poster.  I've tried installing from the most recent master on git and from extensions.gnome.org.  I'm running Gnome Shell 3.16.2 on OpenSUSE Tumbleweed.  The only line from journalctl -xb that looks related to this plugin is this: Gjs-Message: JS LOG: sed -i "s/xxxPATHxxx/\/home\/omod\/\.local\/share\/gnome-shell\/extensions\/cpupower@mko-sl\.de\/cpufreqctl/" /home/omod/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/mko.cpupower.policy

martin31821, 2015-07-20:
yes, that is correct. So did you remove cpufreqctl from /usr/local/bin or whereever you installed it?
cpufreqctl **must not** be installed system-wide as it is executed from within the extensions directory.

I tested it yesterday on fedora, worked like a charm.

@mumblyOMOD - Can you copy  /home/omod/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/mko.cpupower.policy  to /usr/share/polkit-1/actions and then restart gnome-shell?

spacepluk, 2015-07-20:
Arch Linux here with Gnome 3.16.2.  `cpufreqctl` is not in the PATH and mko.cpupower.policy is in place, but it still freezes.  Is there anything I can do to help you debug this?

martin31821, 2015-07-20:
Points you can check:
* mko.cpupower.policy contains the right path to cpufreqctl (should be ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl)
* you can execute `pkexec ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl max 100`, this should work 
* you could try the experimental branch, where I changed some command line handling (should only affect the first installation)

I will try to reproduce this on a bunch of distros

stuaxo, 2015-07-20:
Is there no way it can check for the failure modes and show that it failed in its ui somehow ?

spacepluk, 2015-07-20:
@martin31821 I checked the first two points, and everything seems to be fine.  pkexec asks for the password (via gnome-shell) and then it changes the frequency correctly.

I'll try the experimental branch later and let you know how it goes.

martin31821, 2015-07-20:
@spacepluk thats the problem!

when gnome-shell is starting, pkexec can't create the window to ask you for the password, but blocks the process, which prevents gnome-shell from finishing the startup.
It should execute without any user interaction (i.e. policykit should permit the execution, without asking you)

stuaxo, 2015-07-20:
So I guess it needs to either wait until gnome-shell has started, or just not run at all if it doesn't have permission.

spacepluk, 2015-07-20:
@martin31821 I don't know, for me it also hangs if I install from firefox with gnome-shell already running :?

Arch has polkit 0.112 if that helps.

martin31821, 2015-07-20:
@spacepluk - I assume you flushed the policykit cache (done through restarting polkit.service if you use systemd)?

mumblyOMOD, 2015-07-20:
@martin31821 - I was able to get your extension working by copying /home/omod/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/mko.cpupower.policy to /usr/share/polkit-1/actions manually, at which point the extension would still cause gnome-shell to hang.  I then commented out line 156 [GLib.spawn_command_line_sync('xdg-open ' + EXTENSIONDIR + '/installation');] in extension.js which lets gnome run normally with your extension working perfectly.  So it looks like your extension was hanging on that one line on my system, for some reason.

spacepluk, 2015-07-20:
@mumblyOMOD cool! I just tried that and now it works :) :+1:

martin31821, 2015-07-20:
Please try the latest experimental branch, and confirm the solution.

mumblyOMOD, 2015-07-20:
I removed the extension and the policy from /usr/share/polkit-1/actions, then reinstalled the experimental branch from git.  The experimental branch works perfectly.  Thanks for the quick fix. :+1:

closed 2015-07-20 by martin31821

steelbrain, 2015-07-20:
I can confirm that the fix works <3 <3 <3 <3

spacepluk, 2015-07-20:
great, thank you!

commit 5898b3f4 (2015-07-20): should fix #11