deinstapel/cpupower #12: Installation doubts

marcioAlmada, 2015-06-16:
> If you get the same value as above, everything is working and you can proceed to the next step.

What's the next step? I did all steps, installed the extension and couldn't see any new icon in my toolbar. My model seems compatible:
```
$ ls /sys/devices/system/cpu/intel_pstate/
max_perf_pct  min_perf_pct  no_turbo

fin-ger, 2015-06-17:
Have you installed the extension itself via the gnome extensions website or manually?

marcioAlmada, 2015-06-17:
Via website. Then followed the wiki steps.

martin31821, 2015-06-19:
First you should get the extension from Github, as the latest (important changes) are not yet on extensions.gnome.org, so either you wait some days or you just clone the repository. 
Then you should be able to use the extension (it is possible that you need to restart gnome, press Alt+F2, type r, enter).

marcioAlmada, 2015-06-23:
Still no dice here:

[image]

Unfortunately, no errors messages are logged.

martin31821, 2015-07-18:
As of V2, the installation is much simpler and more intuitive. 
If you install the extension via the gnome website it will check if everything is setup correctly and if not, it will generate all neccessary files and show you instructions how and where to copy them. 

If this doesn't help, I updated the instructions in the wiki.

closed 2015-07-18 by martin31821

marcioAlmada, 2015-07-18:
Removed everything and installed again. No success. The main difference is that now I can't see the extension "settings" button. Looking glass shows no error.

martin31821, 2015-07-21:
please check also the related issues #11 and #15,
You can try the latest git version for the fix.