deinstapel/cpupower #14: Profile extensions

fin-ger, 2015-07-10:
Some cool additional features for profiles would be:

* switch devices on/off (bluetooth, hdd spin down time, etc.)
* set display brightness

ayoungethan, 2020-07-02:
Is this appropriate? It would IMO be "mission creep" and turning the extension into something beyond its original intent (to manage the CPU power and performance parameters) into a general power management interface.

however, IF you are interested in doing this, you might get in touch and coordinate with the Pop!_OS devs, as they are intending to do the same with the System76 power management extension: https://github.com/pop-os/system76-power/issues/128

Also, I've done some background research on power management interface design on sane UX and parameters and posted it here: https://pop-planet.info/forums/threads/power-management-design-interface.739

EJFJCorp, 2021-02-03:
> Some cool additional features for profiles would be:
> 
> * switch devices on/off (bluetooth, hdd spin down time, etc.)
> * set display brightness

Yes, highly requested, done in ErgoSum Operating System, but without any request from user, as automated by real user needs depending on what he is doing (with total protection of any daten of the user, whi cannot forget any object under the chair!). Anything need to be EJFJ Positively Contextual, since March 15th, 1996 at least... and those who are blind need screen to be automatically shut of, except if they can see a bit, and in the case, the screen need to educate each eye to detect ligtings and re-educate the retina... Any problem need to be solved, as soon as possible, as globally as possible, by taking anything in account! Anything evolutes, nothing is totally finished, except this bad old wold, without considerations of human needs who need to adapt to technologies specificities!

fin-ger, 2021-03-15:
@EJFJCorp Happy Birthday! :birthday: :tada:  Although I'm not sure how to help you, as I do not fully understand your comment :disappointed: