deinstapel/cpupower #15: After installation, no icon is shown

Peregring-lk, 2015-07-14:
I've installed your extension following your instructions. Tweak tools shows the extension is successfully installed, and the extension's configuration menu (from tweak tools) only shows me the chance of showing the current CPU freq. But anyway, I see no icons to change CPU governors.

Some data of my laptop: MSI GE60 422XES. Kernel 3.16.0-43, i5-4210H (governors available are only powersave and performance). gnome-shell version 3.10.4. Ubuntu 14.04.

The gnome-shell looking glass shows me no errors.

Peregring-lk, 2015-07-14:
After a while, looking glass throws me the following error:

'_indicator is undefined'.

martin31821, 2015-07-14:
Please paste the output of
`journalctl -xb | grep Gjs`

Peregring-lk, 2015-07-14:
My system (Ubuntu 14.04) doesn't use systemd, but init, so, binary journalctl is not available in my distribution.

What is the equivalent command using init?

Peregring-lk, 2015-07-14:
I hope this helps:

Output of:

$ cd /var/log/gdm/
$ sudo grep -rin "Gjs" .

in pastebin:

http://pastebin.com/80Yv7GYh

So, it seems something is trying to get the non-existent value `org.gnome.SettingsDaemon.Smartcard`. I don't know if that error is generated by your extension or something else, but in my system, the correct entry should be:

`org.gnome.settings-daemon.peripherals.smartcard` (removal-action), or `org.gnome.settings-daemon.plugins.smartcard` (active or priority).

That pastebin is the only error about Gjs I've found in the whole /var/log folder.

martin31821, 2015-07-15:
So this is not generated by my extension, I will see if I can change this

martin31821, 2015-07-19:
Please update to the latest version and check, whether the error still exist. I updated the configuration menu and changed the installation process.

Peregring-lk, 2015-07-22:
Same thing. Before installing from extensions.gnome.com, I've delete every file installed before (the polkit file, the cpufreqctl binary and the extension folder itself). But even more, when I try to install the extension from extensions.gnome.shell, gnome-shell is completly frozen. The extension is installed, but I can't interact with my system, event after restarting. The poolkit file and cpufreqctl is not copied either. Perhaps the GUI is frozen, because it can't copy both files to their corresponding places from the website (/usr/share/polkit-1/actions/ and /usr/local/bin) and thus, can't execute the extension.

What I've done is open a terminal tty (Ctrl+Alt+F1) and copy both files manually from ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/, and, after restarting, gnome-shell runs again, but the extension icon isn't shown. So, same thing as before.

There's other problem in the settings menu. When I open the settings windows, from Tweak Tool or from extensions.gnome.org, I see four sections (Energy Saver, Quiet, Multimedia and High Performance), but all of them shows me the same "screen": Show current frequency (Off by default).

I'm starting to think that perhaps I need some additional javascript libraries. I'm not a javascript programmer and I think I have a minimalist installation (the default one when you install a fresh installation of Ubuntu 14.04, because my laptop is one week old).

Peregring-lk, 2015-07-22:
Is there any way to turn off and on the extension from terminal? If I had a way to "throw" the extension from terminal perhaps I can paste you any reported error.

martin31821, 2015-07-22:
you could try gnome-shell --replace which should redirect any errors/logs to your console.

Peregring-lk, 2015-07-22:
I think nothing related to your extension, but from others:

Gjs-Message: JS LOG: GNOME Shell started at Wed Jul 22 2015 18:09:32 GMT+0200 (CEST)

(gnome-shell:3852): Gjs-WARNING **: JS ERROR: could not get remote objects for service org.gnome.SettingsDaemon.Smartcard path /org/gnome/SettingsDaemon/Smartcard: Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.SettingsDaemon.Smartcard was not provided by any .service files
_proxyInvoker/asyncCallback@resource:///org/gnome/gjs/modules/overrides/Gio.js:83

Gjs-Message: JS LOG: PopupMenu.arrowIcon is not a function
Gjs-Message: JS WARNING: [/home/.../.local/share/gnome-shell/extensions/gmail_notify@jablona123.pl/extension.js 162]: anonymous function does not always return a value
Gjs-Message: JS WARNING: [/home/.../.local/share/gnome-shell/extensions/gmail_notify@jablona123.pl/imap.js 92]: anonymous function does not always return a value
Gjs-Message: JS WARNING: [/home/.../.local/share/gnome-shell/extensions/gmail_notify@jablona123.pl/biginteger.js 97]: function BigInteger does not always return a value
Gjs-Message: JS WARNING: [/home/.../.local/share/gnome-shell/extensions/gmail_notify@jablona123.pl/rexml.js 146]: variable str redeclares argument
Gjs-Message: JS LOG: Init Gmail notify version 0.3.6
Gjs-Message: JS LOG: init error:button is undefined
Gjs-Message: JS LOG:  Enbling Gmail notify version 0.3.6
Gjs-Message: JS LOG: System monitor applet init from /home/.../.local/share/gnome-shell/extensions/system-monitor@paradoxxx.zero.gmail.com
Gjs-Message: JS LOG: System monitor applet enabling
Gjs-Message: JS LOG: System monitor applet enabling done
Gjs-Message: JS LOG: loading default theme (Adwaita)

StevenWolfe, 2015-09-01:
Hello, just found this extension and (think) I had the same issue.  Running Ubuntu Gnome 14.04 LTS with Gnome Shell 3.10.4.  The extension installs from extensions.gnome.org without issue, but there is no pop-up afterwards.

I'm seeing this error when trying to open the settings for the extension via GNOME Tweak Tool:

Error: Requiring GWeather, version none: Typelib file for namespace 'GWeather' (any version) not found

    Stack trace:
      @/home/wolfe/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:36
      Application<._getExtensionPrefsModule@/usr/share/gnome-shell/js/extensionPrefs/main.js:86
      wrapper@resource:///org/gnome/gjs/modules/lang.js:169
      Application<._selectExtension@/usr/share/gnome-shell/js/extensionPrefs/main.js:101
      wrapper@resource:///org/gnome/gjs/modules/lang.js:169
      Application<._extensionSelected@/usr/share/gnome-shell/js/extensionPrefs/main.js:121
      wrapper@resource:///org/gnome/gjs/modules/lang.js:169
      Application<._selectExtension@/usr/share/gnome-shell/js/extensionPrefs/main.js:112
      wrapper@resource:///org/gnome/gjs/modules/lang.js:169
      Application<._extensionsLoaded@/usr/share/gnome-shell/js/extensionPrefs/main.js:221
      wrapper@resource:///org/gnome/gjs/modules/lang.js:169
      _emit@resource:///org/gnome/gjs/modules/signals.js:124
      ExtensionFinder<._extensionsLoaded@/usr/share/gnome-shell/js/misc/extensionUtils.js:178
      wrapper@resource:///org/gnome/gjs/modules/lang.js:169
      done@/usr/share/gnome-shell/js/misc/fileUtils.js:33
      _collectFromDirectoryAsync/</<@/usr/share/gnome-shell/js/misc/fileUtils.js:51
      onNextFileComplete@/usr/share/gnome-shell/js/misc/fileUtils.js:21
      main@/usr/share/gnome-shell/js/extensionPrefs/main.js:278
      @<command line>:1

I'm also seeing the errors re: Smartcard with `sudo grep -rin "Gjs" /var/log/gdm/`

martin31821, 2015-09-01:
See #16 for this

teamblubee, 2015-11-09:
I had these issues but on debian 8 installing this package fixed the issue: libgweather-3-dev

martin31821, 2015-11-24:
The gweather dependency was removed some time ago (#16).

closed 2015-11-24 by martin31821