deinstapel/cpupower #17: Can not set frequency after suspending

martin31821, 2015-07-27:
After suspending, the extension does not retain the configuration, and loses the ability to change de CPU configuration. If I choose a profile, the extension apparently switches to that profile, but frequency never changes.

Version information:

    Shell version: 3.14.4
    Extension version: 4

martin31821, 2015-07-27:
Can you try to update the frequency manually?
so, as root:
`echo 100 > /sys/devices/system/cpu/intel_pstate/max_perf_pct`
`echo 100 > /sys/devices/system/cpu/intel_pstate/min_perf_pct`

then read out the values (using cat) and post the result

caprilo, 2015-07-27:
I entered the commands and the output with cat is

    100
    100

I allso tried with `echo 30` and the result was fine. The extension displayed the change.

martin31821, 2015-08-02:
I could reproduce this once, but in my case I could not set the frequency via command line, neither via the extension itself so I think this is a problem related to the pstate driver. In case this happens, you can just suspend and wake up again so the pstate driver is resetted.

cyberalex4life, 2015-09-20:
just use cpupower to set governor, but for global options you should mask cpupower.service (if on systemd), then use tlp. tlp has settings for intel_pstate. Onced masked, cpupower does not start anymore. tlp can help you set max and min frequencies

closed 2016-03-23 by martin31821