deinstapel/cpupower #18: Pull request: Save last settings and apply them on startup and resume. Governor change option.

cyberalex4life, 2015-09-26:
I think this extension should be able to save last applied settings and apply them on resume after suspend or restart/startup. Some settings are overridden by cpupower.service or by tlp.service (if used) on startup. I've been able to disable cpupower and mask it, to let tlp apply my settings. I've set in tlp the procentages I want by default and the governor for AC or BAT. I think the governor would be a new nice feature to add to this extensions.

martin31821, 2015-09-26:
If you use cpupower.service you can not set the governor as it can only read the frequency.
For me, `cpupower frequency-info` shows that it doesn't actually control the frequency of the cpu:
`CPUs which need to have their frequency coordinated by software: 0`
I never used tlp but it should work as drop-in-replacement for cpupower.

I try to implement the last applied settings and reset them on startup

closed 2016-04-23 by martin31821

cyberalex4life, 2016-04-23:
Thanks for this enhancement. Sorry for:
Feature Request: Posting Error #31,

commit 3b3cf192 (2016-04-23): implement frequency restore on startup, fix #18