deinstapel/cpupower #21: cpufreqctl min_perf_pct: No such file or directory in VMWare emulated box

MikhailRokhin, 2015-12-22:
.local/bin/cpufreqctl: line 54: /sys/devices/system/cpu/intel_pstate/min_perf_pct: No such file or directory

/sys/devices/system/cpu/
/sys/devices/system/cpu/cpu0
/sys/devices/system/cpu/cpu0/cache
/sys/devices/system/cpu/cpu0/cache/power/control
/sys/devices/system/cpu/cpu0/cache/power/async
/sys/devices/system/cpu/cpu0/cache/power/runtime_active_kids
/sys/devices/system/cpu/cpu0/cache/power/runtime_active_time
/sys/devices/system/cpu/cpu0/cache/index0/power/control
/sys/devices/system/cpu/cpu0/cache/index0/power/async
/sys/devices/system/cpu/cpu0/cache/index0/power/runtime_active_kids
/sys/devices/system/cpu/cpu0/cache/index0/power/runtime_active_time
/sys/devices/system/cpu/cpu0/cache/index0/shared_cpu_map
/sys/devices/system/cpu/cpu0/cache/index0/shared_cpu_list
/sys/devices/system/cpu/cpu0/cache/index0/coherency_line_size
/sys/devices/system/cpu/cpu0/cache/index0/physical_line_partition
/sys/devices/system/cpu/cpu0/cache/index0/ways_of_associativity
/sys/devices/system/cpu/cpu0/cache/index1/power/control
/sys/devices/system/cpu/cpu0/cache/index1/power/async
/sys/devices/system/cpu/cpu0/cache/index1/power/runtime_active_kids
/sys/devices/system/cpu/cpu0/cache/index1/power/runtime_active_time
/sys/devices/system/cpu/cpu0/cache/index1/shared_cpu_map
/sys/devices/system/cpu/cpu0/cache/index1/shared_cpu_list
/sys/devices/system/cpu/cpu0/cache/index1/coherency_line_size
/sys/devices/system/cpu/cpu0/cache/index1/physical_line_partition
/sys/devices/system/cpu/cpu0/cache/index1/ways_of_associativity
/sys/devices/system/cpu/cpu0/cache/index2/power/control
/sys/devices/system/cpu/cpu0/cache/index2/power/async
/sys/devices/system/cpu/cpu0/cache/index2/power/runtime_active_kids
/sys/devices/system/cpu/cpu0/cache/index2/power/runtime_active_time
/sys/devices/system/cpu/cpu0/cache/index2/shared_cpu_map
/sys/devices/system/cpu/cpu0/cache/index2/shared_cpu_list
/sys/devices/system/cpu/cpu0/cache/index2/coherency_line_size
/sys/devices/system/cpu/cpu0/cache/index2/physical_line_partition
/sys/devices/system/cpu/cpu0/cache/index2/ways_of_associativity
/sys/devices/system/cpu/cpu0/cache/index3/power/control
/sys/devices/system/cpu/cpu0/cache/index3/power/async
/sys/devices/system/cpu/cpu0/cache/index3/power/runtime_active_kids
/sys/devices/system/cpu/cpu0/cache/index3/power/runtime_active_time
/sys/devices/system/cpu/cpu0/cache/index3/shared_cpu_map
/sys/devices/system/cpu/cpu0/cache/index3/shared_cpu_list
/sys/devices/system/cpu/cpu0/cache/index3/coherency_line_size
/sys/devices/system/cpu/cpu0/cache/index3/physical_line_partition
/sys/devices/system/cpu/cpu0/cache/index3/ways_of_associativity
/sys/devices/system/cpu/cpu0/power/control
/sys/devices/system/cpu/cpu0/power/async
/sys/devices/system/cpu/cpu0/power/runtime_active_kids
/sys/devices/system/cpu/cpu0/power/runtime_active_time
/sys/devices/system/cpu/cpu0/crash_notes
/sys/devices/system/cpu/cpu0/crash_notes_size
/sys/devices/system/cpu/cpu0/topology/core_id
/sys/devices/system/cpu/cpu0/topology/core_siblings
/sys/devices/system/cpu/cpu0/topology/core_siblings_list
/sys/devices/system/cpu/cpu0/topology/physical_package_id
/sys/devices/system/cpu/cpu0/microcode
/sys/devices/system/cpu/cpu0/microcode/processor_flags
/sys/devices/system/cpu/cpu1
/sys/devices/system/cpu/cpu1/cache
/sys/devices/system/cpu/cpu1/cache/power/control
/sys/devices/system/cpu/cpu1/cache/power/async
/sys/devices/system/cpu/cpu1/cache/power/runtime_active_kids
/sys/devices/system/cpu/cpu1/cache/power/runtime_active_time
/sys/devices/system/cpu/cpu1/cache/index0/power/control
/sys/devices/system/cpu/cpu1/cache/index0/power/async
/sys/devices/system/cpu/cpu1/cache/index0/power/runtime_active_kids
/sys/devices/system/cpu/cpu1/cache/index0/power/runtime_active_time
/sys/devices/system/cpu/cpu1/cache/index0/shared_cpu_map
/sys/devices/system/cpu/cpu1/cache/index0/shared_cpu_list
/sys/devices/system/cpu/cpu1/cache/index0/coherency_line_size
/sys/devices/system/cpu/cpu1/cache/index0/physical_line_partition
/sys/devices/system/cpu/cpu1/cache/index0/ways_of_associativity
/sys/devices/system/cpu/cpu1/cache/index1/power/control
/sys/devices/system/cpu/cpu1/cache/index1/power/async
/sys/devices/system/cpu/cpu1/cache/index1/power/runtime_active_kids
/sys/devices/system/cpu/cpu1/cache/index1/power/runtime_active_time
/sys/devices/system/cpu/cpu1/cache/index1/shared_cpu_map
/sys/devices/system/cpu/cpu1/cache/index1/shared_cpu_list
/sys/devices/system/cpu/cpu1/cache/index1/coherency_line_size
/sys/devices/system/cpu/cpu1/cache/index1/physical_line_partition
/sys/devices/system/cpu/cpu1/cache/index1/ways_of_associativity
/sys/devices/system/cpu/cpu1/cache/index2/power/control
/sys/devices/system/cpu/cpu1/cache/index2/power/async
/sys/devices/system/cpu/cpu1/cache/index2/power/runtime_active_kids
/sys/devices/system/cpu/cpu1/cache/index2/power/runtime_active_time
/sys/devices/system/cpu/cpu1/cache/index2/shared_cpu_map
/sys/devices/system/cpu/cpu1/cache/index2/shared_cpu_list
/sys/devices/system/cpu/cpu1/cache/index2/coherency_line_size
/sys/devices/system/cpu/cpu1/cache/index2/physical_line_partition
/sys/devices/system/cpu/cpu1/cache/index2/ways_of_associativity
/sys/devices/system/cpu/cpu1/cache/index3/power/control
/sys/devices/system/cpu/cpu1/cache/index3/power/async
/sys/devices/system/cpu/cpu1/cache/index3/power/runtime_active_kids
/sys/devices/system/cpu/cpu1/cache/index3/power/runtime_active_time
/sys/devices/system/cpu/cpu1/cache/index3/shared_cpu_map
/sys/devices/system/cpu/cpu1/cache/index3/shared_cpu_list
/sys/devices/system/cpu/cpu1/cache/index3/coherency_line_size
/sys/devices/system/cpu/cpu1/cache/index3/physical_line_partition
/sys/devices/system/cpu/cpu1/cache/index3/ways_of_associativity
/sys/devices/system/cpu/cpu1/power/control
/sys/devices/system/cpu/cpu1/power/async
/sys/devices/system/cpu/cpu1/power/runtime_active_kids
/sys/devices/system/cpu/cpu1/power/runtime_active_time
/sys/devices/system/cpu/cpu1/crash_notes
/sys/devices/system/cpu/cpu1/crash_notes_size
/sys/devices/system/cpu/cpu1/topology/core_id
/sys/devices/system/cpu/cpu1/topology/core_siblings
/sys/devices/system/cpu/cpu1/topology/core_siblings_list
/sys/devices/system/cpu/cpu1/topology/physical_package_id
/sys/devices/system/cpu/cpu1/microcode
/sys/devices/system/cpu/cpu1/microcode/processor_flags
/sys/devices/system/cpu/cpuidle
/sys/devices/system/cpu/cpuidle/current_driver
/sys/devices/system/cpu/cpuidle/current_governor_ro
/sys/devices/system/cpu/power/control
/sys/devices/system/cpu/power/async
/sys/devices/system/cpu/power/runtime_active_kids
/sys/devices/system/cpu/power/runtime_active_time
/sys/devices/system/cpu/microcode

MikhailRokhin, 2015-12-22:
Bad output of freq in task bar

[image]

martin31821, 2015-12-22:
You need the intel_pstate driver to make it work, which is unavailable in any kind of virtual machine.

closed 2015-12-22 by martin31821