deinstapel/cpupower #24: Show active choosen profile

xdajog, 2016-03-10:
#### Request for enhancement

It would be very nice if one could see which profile is currently selected (maybe a simple check mark on the last selected profile or so)

martin31821, 2016-03-12:
How can I handle the case if the user selects a profile and changes any setting afterwards?
Just show the last set profile or don't highlight any profile at all?

xdajog, 2016-03-13:
For me this would be clear enough and logical:

if the user changes the sliders (I think that is what you meant) then it shouldn't highlight any profile at all. The reason behind is that it **is** actually **no** profile anymore. When the user wants to use those settings again he/she should save a new profile for this..

martin31821, 2016-03-23:
I created a sample implementation for testing purposes.
I think it will go into the next release.

xdajog, 2016-03-25:
Looks really great !! Thanks for implementing!

closed 2016-03-25 by martin31821