deinstapel/cpupower #28: Frequency Does Not Change

dsreyes1014, 2016-04-21:
After switching between profiles for the cpu frequencies the frequency number does not change.

martin31821, 2016-04-21:
Can you provide more information?

Which CPU model, which gnome version, which extension version etc.
Please also include logs (journald, extension log etc.)

dsreyes1014, 2016-04-22:
I'm running an i5 4570t Haswell cpu on Gentoo Linux kernel 4.5.1, Gnome 3.18 not sure about extension version as there's no version number that comes with it so I'm assuming latest git through extensions.gnome.org.

martin31821, 2016-04-22:
Your CPU is supported, Gnome 3.18 should work as well.

Can you try to set the frequency manually using the pstate files and cpufreqctl?
please take a look at #6 and #17 as this seems to be related.

make sure your policykit action is working correctly [https://github.com/martin31821/cpupower/wiki/Installation](Wiki Installation)

dsreyes1014, 2016-04-23:
the link to cpupower is a dead link.  I'll try it manually and see if it works.

martin31821, 2016-04-23:
Oops, got the wrong order in markdown. Fixed it.

dsreyes1014, 2016-04-23:
Hey martin after seeing #17 I noticed I didn't compile intel pstate driver in my kernel build.  Very sorry I missed.  It's working now as it should it seems and all is well.

closed 2016-04-23 by martin31821