deinstapel/cpupower #29: Option to put the extention in system menu

qmarlats, 2016-04-22:
Can you add an option to place the extention in top right menu please?

martin31821, 2016-04-22:
What do you mean by "top right menu" exactly? 
I think it's the status window where volume etc. are controlled?

If so, I can try. 
Do you want the profile selection and/or the sliders / switches?

qmarlats, 2016-04-22:
Yep it's this menu, sorry I don't know how it's called. ^^ I think you can simply add an option to move the extension in this menu, by replacing the dropdown by the sort of dropdown in the right menu (like this: https://deblinux.files.wordpress.com/2014/06/media-player-indicator.png?w=665). I hope you understood me, I'm not clear at all and I'm French ^^
[image]

shihrer, 2016-05-20:
I think this would be a great option to have.

tonylambiris, 2018-04-19:
I like the idea, there's an extension I use that does something similar  I'll keep it on the back-burner.

KristiaN1337, 2020-01-25:
A friend and I have been trying to make this work for the last couple of days, we're not so good, obviously, but could we at least get some pointers on what we'd have to do if this feature will never arrive?

Thanks

martin31821, 2020-01-25:
Sorry, but I don't know how to achieve this. You might want to have a look at the Bluetooth Quick Connect or Sound Input & Output Device Chooser extensions, which both integrate into the system menu. 

If anyone wants to prepare a PR, I'd be happy to merge this, but I'd like to keep this functionality behind a config option.