deinstapel/cpupower #30: Previously Selected Profile Isn't Shown To Be Selected After Reopening

dsreyes1014, 2016-04-23:
I'm losing the bullet point next to the profile.  I'm noticing that the bullet point when selecting a profile is missing after reopening the extension from the top bar.  So it's a little tough to remember which profile I'm on after awhile.  I'm on Gnome 3.18 Gentoo kernel 4.5.2.

martin31821, 2016-04-23:
The bullet is shown when the settings (min, max, turbo boost) match the settings of any profile.

I suppose this only happens when on profile quiet or energy saver? 
If so, it's a problem with your minimum frequency (see #25)

dsreyes1014, 2016-04-23:
Ahh you're right.  I setup the profiles to match my lowest frequency and it stays on when reopening.  Thanks for Martin!!!

closed 2016-04-24 by martin31821