deinstapel/cpupower #32: Coming Back From Suspend, Freq Number Doesn't change

dsreyes1014, 2016-04-23:
When coming back on from suspend the frequency number doesn't change no matter what profile I use.

martin31821, 2016-04-27:
Is this reproducable?
Can you try to set the frequency manually using cpufreqctl?

Znert, 2016-05-01:
I am not sure how to reproduce this error, but I have the same issue. 

I noticed however, that the governor of my CPU is always set to "performance" (not only after suspend but also when plugging the laptop in) which does not allow frequency changes. 
Apparently this is intended behavior, however, as it is less efficient to calculate how high the frequency should be for the next task. 

So I don't think that it's the fault of the extension but rather the kernel's. 

A workaround is to set the governor back to powersave via "sudo cpufreq-set -g powersave" (or any other governor which allows freqency changes, check "cpufreq-info" for more details for your CPU).

Not sure if it's helpful or not, but my CPU is Intel i5-4210U.

martin31821, 2016-05-01:
When using intel pstate, you shouldn't use any kind of governor as this is the oldfashioned acpi power management. 
Please check if you are actually using the pstate driver to manage your frequency.

Znert, 2016-05-02:
Yesterday I updated my laptop to Ubuntu Gnome 16.04, which did not go down smoothly at all and so I had to reinstall Ubuntu as a whole... 

To good thing is, that the problem no longer persist and I can freely change the frequency at any time.

The bad thing however, I no longer can reproduce the error myself and also can't double check if I was using psate or not. But still, I am convinced that I actually was using intel pstate according to "cpufreq-info"...

closed 2017-05-14 by martin31821