deinstapel/cpupower #35: A question

ilkerhk, 2016-06-11:
Hi, enjoying your extension for a while. 
The only thing I wish it had is to set default profiles in battery and AC modes.  

Does anyone knows an easy way to achieve this (preferably without dbus monitoring) ? 

Currently I am running below commands in gdm startup. Would like to trigger scripts when adapter connect/disconnect...
thanks....

pkexec ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl min 0
pkexec ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl max 90
pkexec ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl turbo 0

martin31821, 2016-06-11:
I think the only way to do this is monitoring the d-bus, so I think you would have to create something like this. 

See #4 as related issue, dbus-integration was planned some time ago.

fin-ger, 2017-04-29:
@ilkerhk You might want to change the `cpufreqctl` path in your scripts with the upcoming version to `src/cpufreqctl`.

Support for battery/ac-mode profiles will be added in a later version.

sscarduzio, 2020-01-13:
My solution was intercepting the ACPI events for AC adapter connected or disconnected. This is what I did.
Copy cpufreqctl script from this repo and place it under /etc/acpi/, make it executable, and then create these two events in /etc/acpi/events:

```
sscarduzio@nash:/etc/acpi$ cat events/ac_power_attached 
event=ac_adapter ACPI0003:00 00000080 00000001
action=logger "AC ADAPTER IS ATTACHED, moving CPU governor to performance evt:  %e" && /etc/acpi/cpufreqctl turbo 1 && /etc/acpi/cpufreqctl min 50 && /etc/acpi/cpufreqctl max 100

sscarduzio@nash:/etc/acpi$ cat events/ac_power_removed 
event=ac_adapter ACPI0003:00 00000080 00000000
action=logger "AC ADAPTER IS REMOVED, moving CPU governor to powersave evt: %e" && /etc/acpi/cpufreqctl turbo 0 && /etc/acpi/cpufreqctl min 10 && /etc/acpi/cpufreqctl max 50
``

mastercaution, 2020-02-08:
You can modify the [`_onPowerChanged()`](https://github.com/martin31821/cpupower/blob/9550e3abc267049db477add4804c802e0deb0a4b/src/indicator.js#L100) function src/indicator.js . 
There you can either hard code the values like @sscarduzio dit it or switch to a specific profile by using the [`_applyProfile(profile)`](https://github.com/martin31821/cpupower/blob/9550e3abc267049db477add4804c802e0deb0a4b/src/indicator.js#L197) function.

### Hard Code (example of ac removed)
```js
else if (new_state === UPower.DeviceState.DISCHARGING)
{
    global.log ("Power state changed: discharging");
    
    // set min to 10%
    this.minVal = 10;
    this._updateMin();

   // set max to 50%
    this.maxVal = 50;
    this._updateMax();

   // turn of turbo boost
    this.isTurboBoostActive = false;
    this._updateTurbo();

    // update UI to see the changes
    this._updateUi();
}
```
I didn't test this code itself but with this procedure the UI should update the changes.

### Switch Profile
In order to automatically switch profiles, the easiest way is probably to search for a specific profile name (e.g. [Auto] Battery) and apply it with `_applyProfile(profile)`:
```js
else if (new_state === UPower.DeviceState.DISCHARGING)
{
    global.log ("Power state changed: discharging");

    // run through all profiles
    for(var i = 0; i < this.profiles.length; i++)
    {
        // find the profile named "[Auto] Battery"
        if (this.profiles[i].Profile.Name == '[Auto] Battery')
        {
            // apply profile
            this._applyProfile(this.profiles[i].Profile);
            break;
        }
    }
}
```
I'm using this approach so that I can change the automatic profiles on the fly.

You'll find the indicator.js in `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/`. After you made the modifications, you need to restart gnome-shell i guess (simply restart your computer).

martin31821, 2020-02-08:
If you could prepare a PR for this, I'd be happy to merge, great work!

mastercaution, 2020-02-08:
Yes of course!
I would implement a little option in the preferences to enable or disable auto-switching along with the possibility to choose the profile to switch to, because I think it's nice if it's configurable through the UI ;)

I need some days to work on this and test it.
Thanks for asking for the PR!

fin-ger, 2020-03-31:
This got implemented in #96

closed 2020-03-31 by fin-ger