deinstapel/cpupower #36: Max Freq - settings are not honored

postadelmaga, 2016-07-14:
I have correctly installed the extension and it is working so far however if I run some up that use a lot of cpu the max limit is automatically changed.

I'm not sure if this happens because of some kind of HW controller on the frequency. ( I'm not running  `thermald` or other similar software )

**OS: antergos**
**CPU: Sandybridge ( Intel(R) Core(TM) i5-2430M CPU @ 2.40GHz) **

martin31821, 2016-07-14:
Can you try and set the frequency manually using the shell script which is bundled within this extension?
I think some other process (maybe your BIOS/EFI as well) is overwriting the cpu frequency limit you set.

Additionally I don't know exactly if Sandybridge is alreadyy supported with the pstate driver or if the values are readonly.

postadelmaga, 2016-07-14:
I tried both with the script `cpufreqctl` and direct on sysfs:

> echo 1 > /sys/devices/system/cpu/intel_pstate/no_turbo
> echo 800 > /sys/kernel/debug/dri/0/i915_max_freq  # GPU
> echo 40 > /sys/devices/system/cpu/intel_pstate/max_perf_pct

,,,  and also disabling `Intel SpeedStep` in the bios but so far the limits under heavy load the limit are automatically changed ( probably by the BIOS as you point out ) 

I think the issue is the bios so you can probably close this bug.

Anyway maybe you could implement the GPU limit as well I guess it would be a good adding to the extension/script

closed 2016-07-14 by martin31821