deinstapel/cpupower #37: Blank name results in error in overview

dnivi3, 2016-09-26:
Everything works correctly, but I accidentally added an entry without a blank name. This resulted in an error in which metadata information about the package is displayed instead of the name of the mode. Please see the attached screenshot. 

[image]

martin31821, 2016-09-26:
Got it reproduced, I will look into it.

closed 2016-09-26 by martin31821

dnivi3, 2016-09-26:
Thanks for fixing this! I didn't realise I actually had an empty item in the list, so I didn't think of that I could delete it. 

As a thank, do you happen to have a Bitcoin-address I can send a coffee to?

martin31821, 2016-09-26:
Please send me an email, my address can be found in my profile.

commit cd925ae5 (2016-09-26): Fix #37

Turns out this was a bug/strange behaviour in gettext
which returns metainformation about the loaded translation
when attempting to translate the empty string.

More information can be found here:
http://stackoverflow.com/questions/7610144/why-does-gettext-translate-an-empty-string-to-the-po-header-text