deinstapel/cpupower #38: Latest update is botched

Vistaus, 2016-10-08:
This extension was working fine on my Arch Linux with GNOME 3.22 but the latest update from the GNOME Extensions website results in a complete hang of the entire system and I have to manually rm the extension folder from a tty to have a working system again.

Could you please fix this?

martin31821, 2016-10-08:
I'll look into it tomorrow, I think I need to setup an ubuntu to reproduce this, it is related to glib.

Vistaus, 2016-10-21:
Are you still looking into this? 'Cause I miss this extension! :cry: Btw, I'm using Arch, not Ubuntu.

fin-ger, 2016-10-22:
How can I install the v7 tagged version without extensions.gnome.org but with a git checkout? I have a gentoo running and can test multiple configurations of glib. I will try to debug the issue.

martin31821, 2016-10-23:
I haven't been able to reproduce this on a bunch of distros (Ubuntu Gnome, Arch, Gentoo, Fedora)

You could try and delete the whole extension installation folder, i.e. do a clean install to prevent update problems.

Vistaus, 2016-10-23:
Tried that already when I submitted this issue but it didn't work. But it does work now. The only problem is that, after installing and restarting Shell, it says "Installation required" even though I've followed all of the instructions (copying the cpufreq file and changing permissions and stuff).

martin31821, 2016-10-23:
The file which is used to determine if the installation was successful is `.pgen` in the extension directory.
So running `touch $EXTENSIONDIR/.pgen` should do the job.

Vistaus, 2016-10-24:
Thanks. Got a little further now but it's still not working yet 'cause when I click "High Performance" it's asking for my sudo password but it doesn't accept it. My sudo password works fine elsewhere, for example in a terminal window.

What to do now?

martin31821, 2016-10-24:
Can you post the contents of your `mko.cpupower.policy` file here?
Please also check if the cpufreqctl executable matches the path of `mko.cpupower.policy`

martin31821, 2017-02-13:
@Vistaus FYI I have done a clean install yesterday on arch (just git clone and follow the installation instructions) and it worked like a charm. I'm using Archlinux on an i7-6820HK, so I'm closing this for now. Feel free to open a new ticket when the issue persists.

closed 2017-02-13 by martin31821