deinstapel/cpupower #43: policy-kit installation wrong?!

steadfasterX, 2017-05-08:
In my case the following happened:

I installed the extension over extensions.gnome.org
Then after the extension gets enabled gnome asks multiple times for my password to execute _cpufreqctl_

The policy file gets installed correctly into:
_/usr/share/polkit-1/actions/mko.cpupower.policy_

but within that file I find the line:

`<annotate key="org.freedesktop.policykit.exec.path">/home/tfi/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl</annotate>`

which is either not correct or the installer missed to moved the cpufreqctl file from the directory src/.

I "fixed" it by modifying the path to:
`<annotate key="org.freedesktop.policykit.exec.path">/home/tfi/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/cpufreqctl</annotate>`

steadfasterX, 2017-05-08:
btw: `/usr/share/polkit-1/actions/mko.cpupower.setcpufreq.policy` contains the path to **src/**

martin31821, 2017-05-08:
this could be indeed a leftover from a previous version.
I will adapt the installer to remove old versions of the policy file...

Oliph, 2017-07-26:
I add the same issue and the solution given here solve the issue for me.

superkartoffel, 2017-08-08:
I can confirm this issues. I also had a second policy, probably from a previous version and got ask to enter a password multiple times on each login and whenever I tried to use the extension.
Removing the second policy resolved the issue for me.

eode, 2017-08-30:
Still broken.  Still the same issue.  Two polkit files, the older being `mko.cpupower.policy`, the newer being `mko.cpupower.setcpufreq.policy`.  Deleting the older one fixes the issue.

fin-ger, 2020-03-31:
This is being worked in in #107

closed 2020-04-14 by fin-ger