deinstapel/cpupower #44: Option to not restore last setting

biji, 2017-05-14:
In my case before suspending , I use high performance profile on AC, then after resuming my system, I use slower cpu setting, set by TLP, but cpupower set it back to high profile. Please add option to not restoring last setting

Thanks

martin31821, 2017-05-14:
Good point, will implement this.

levinit, 2017-08-23:
support this proposal.Now I use ['cpupower'](https://wiki.archlinux.org/index.php/CPU_frequency_scaling#cpupower)to keep turbo boost off,but it is not convenient for using.

tonylambiris, 2018-04-19:
@martin31821 I can take a crack at this -- I like the idea I just need any pointers/quick-starts regarding how to build out more UI elements. I tried Builder and Glide but I didn't see any intuitive hints in the UI.

What app/IDE do you use for creating widgets and binding actions? And also if you can kick me a few curated bookmarks you deemed save-worthy please kick em my way (everything I've been reading isn't what I want or it's super out of date).

mastercaution, 2020-02-29:
> In my case before suspending , I use high performance profile on AC, then after resuming my system, I use slower cpu setting, set by TLP, but cpupower set it back to high profile.

This is exactly why I implemented #96. Now TLP don't need to touch the pstate values anymore instead the extension can handle the AC and battery settings.

> What app/IDE do you use for creating widgets and binding actions?

I used Glade in case you are still interested.

fin-ger, 2020-04-14:
@biji is this issue fixed?

fin-ger, 2020-04-21:
If this is still an issue, please reopen!

closed 2020-04-21 by fin-ger