deinstapel/cpupower #52: Shell freezes every 2 seconds

timoschwarzer, 2017-10-29:
The entire shell freezes every 2 seconds for about 100-300 ms.

GNOME Shell 3.26.1
Extension as of https://github.com/martin31821/cpupower/commit/295f38be4f0ac36498a96d8d8854071fdc47fedc
CPU: i7 6700K
Kernel: 4.13.0-16-generic

martin31821, 2017-10-29:
Can you try to disable `Show current frequency` in the settings and check if the freezes are still present?

Looks like a performance problem to me.

timoschwarzer, 2017-10-29:
Freezes are still present with the setting turned off.
It worked fine on Ubuntu GNOME 17.04. I did a fresh install of Ubuntu 17.10 yesterday and noticed the freezes.
And it seems like I'm not the only one experiencing this: https://extensions.gnome.org/extension/945/cpu-power-manager/ (3rd comment)
But I can't find the reason for the freezes...

martin31821, 2017-10-29:
I'll setup an ubuntu 17.10 to reproduce this issue.

martin31821, 2017-11-03:
Ok so I got the problem reproduced on my machine, but in a much more strange way.
If I activate the extension I can only use gnome shell once per 5 seconds.

@tonylambiris I tracked it down to the new way of calculating CPU frequencies which is very slow due to many file opens for many cores (for me having 40 threads it takes aroung 4 seconds)
Maybe you can look at improving the performance?

tonylambiris, 2017-11-04:
@martin31821 I'll poke around.

timoschwarzer, 2017-11-04:
@martin31821 I have 4 physical cores, 8 virtual cores with Hyper Threading.

tonylambiris, 2017-11-04:
https://github.com/martin31821/cpupower/pull/54

martin31821, 2017-11-09:
Fixed by #54

closed 2017-11-09 by martin31821

timoschwarzer, 2017-11-12:
@martin31821 Is this bug fix pending on extenstions.gnome.org?

martin31821, 2017-11-12:
Will release it tomorrow

fin-ger, 2021-05-03:
I'm currently rewriting the `cpufreqctl` tool to support other scaling drivers than `intel_pstate`. I'm currently looking into integrating the frequency polling into the tool, to make it interchangeable with the scaling-driver backend. In order to avoid hitting this issue again, I would like to ask someone with more CPU cores to test my current approach (I tested it with 12 cores):

```sh
for file in /sys/devices/system/cpu/cpufreq/policy*/scaling_cur_freq; do cat "$file"& done | sort -n | awk 'NR==1{printf "min: %d\n", $1};{sum+=$1}END{printf "max: %d\n", $1; printf "avg: %.0f\n", sum/NR}'
```
(sorry for the horrible one-liner)

Explanation of code:

The code reads the `/sys/devices/system/cpu/cpufreq/policy[0-n]/scaling_cur_freq` files where the current frequency of each CPU core is stored. The `sort` and `awk` are used to gather the minimum, maximum, and average CPU frequency. The `cat` is send to background to read all the frequencies of all cores in parallel. The pipe to `sort` collects all the lines emitted by the parallel running `cat` and sorts them numerically.

Thanks for any feedback!