deinstapel/cpupower #57: Extension does not appear in GNOME Shell 3.18.5

tommyhlw, 2017-12-18:
For no reason the widget does not appear anymore after turning on on the shell extensions website.

- pstate is enabled
> $ ls /sys/devices/system/cpu/cpu0/cpufreq/ 
> affected_cpus              
 related_cpus               
  scaling_max_freq
> cpuinfo_cur_freq          
  scaling_available_governors  
scaling_min_freq
> cpuinfo_max_freq           
 scaling_cur_freq            
 scaling_setspeed
> cpuinfo_min_freq          
  scaling_driver
> cpuinfo_transition_latency  
scaling_governor
- Intel Core i5-4200U CPU Haswell mobile
- Ubuntu 16.04.3 LTS
- Kernel: 4.4.0-104-generic
- policykit 0.105-2ubuntu2
- GNOME Shell 3.18.5

>$ journalctl -xb | grep cpupower@mko-sl.de
Dez 18 04:31:17 tommy-dell gnome-session[2103]: Gjs-Message: JS LOG: sed -i "s/xxxPATHxxx/\/home\/tommy\/\.local\/share\/gnome-shell\/extensions\/cpupower@mko-sl\.de\/cpufreqctl/" /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/mko.cpupower.policy
Dez 18 04:31:17 tommy-dell gnome-session[2103]: Gjs-Message: JS LOG: Kindprozess »/home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/cpufreqctl« konnte nicht ausgeführt werden (Keine Berechtigung)
Dez 18 04:32:14 tommy-dell gnome-session[2103]: Gjs-Message: JS LOG: Extension "cpupower@mko-sl.de" had error: TypeError: _indicator is undefined
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/sr@latin/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/cs_CZ.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/installer.sh
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/en_US.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/unsupported.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/ru/LC_MESSAGES/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/icons/cpu.svg
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/de_DE.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/de/LC_MESSAGES/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/convenience.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/sr@latin/LC_MESSAGES/gnome-shell-extension-cpupower.mo
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/fr_FR.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/en/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/pl/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/sr/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profilebutton.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/ru_RU.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/notinstalled.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/pl_PL.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/fr/LC_MESSAGES/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/sr_RS.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/mko.cpupower.policy.in
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/de/LC_MESSAGES/gnome-shell-extension-cpupower.mo
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/sr/LC_MESSAGES/gnome-shell-extension-cpupower.mo
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/cs/LC_MESSAGES/gnome-shell-extension-cpupower.mo
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/schemas/org.gnome.shell.extensions.cpupower.gschema.xml
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/ru/LC_MESSAGES/gnome-shell-extension-cpupower.mo
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profile.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/LICENSE
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/pl/LC_MESSAGES/gnome-shell-extension-cpupower.mo
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/cpufreqctl
Dez 18 04:32:20 tommy-dell gnome-session[2103]:    creating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/en/LC_MESSAGES/
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/schemas/gschemas.compiled
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/sr_RS@latin.po
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/template.pot
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js
Dez 18 04:32:20 tommy-dell gnome-session[2103]:   inflating: /home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/locale/en/LC_MESSAGES/gnome-shell-extension-cpupower.mo
[... 37 lines cut ...]
Dez 18 04:51:58 tommy-dell gnome-session[2103]: CPUPowerPrefsWidget<.refreshUI@/home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:519
Dez 18 04:51:58 tommy-dell gnome-session[2103]: CPUPowerPrefsWidget<.loadConfig/<@/home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:538
Dez 18 04:51:58 tommy-dell gnome-session[2103]: CPUPowerPrefsWidget<.profiles@/home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:592
[... cut ...]
Dez 18 04:52:11 tommy-dell gnome-session[2103]: CPUPowerPrefsWidget<.addSwitch/<@/home/tommy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:293

# Creating a new issue

* Check if another issue with the same problem exists
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
  * policykit working and rule working
* Check if the cpufreqctl utility works correctly
* Please include the following information (only for bugreports):
  * CPU Model
  * Linux kernel version
  * policykit version
  * Gnome shell version
  * journalctl -xb (maybe pastebin/filtered)

martin31821, 2017-12-19:
Seems like the installer is not executable. I'll look into it

fin-ger, 2020-04-05:
As gnome 3.18 is EOL, I will close this issue. As of #107, the extension works with Ubuntu 18.04, which is the current LTS of Ubuntu. Please reopen if this is still a bug!

closed 2020-04-05 by fin-ger