deinstapel/cpupower #58: Feature suggestion: Complete uninstallation

jebeld17, 2018-02-16:
Please add a 'Completely Uninstall' option to the CPU Power Manager applet that completely reverses any changes the applet may have done to the user's computer including remove and reverse the scripts installed by the applet to work properly.

This gives users peace of mind that in case the applet does not work for their needs, they can still reverse any changes, more than likely, that may have incurred.

GXGOW, 2018-02-21:
According to #48, you only have to remove one file. It would be helpful to put that in the README though.

martin31821, 2018-02-21:
I'll put it in the readme.

jebeld17, 2018-02-28:
What about to undo the changes?

kiview, 2019-01-03:
I assume it would be reset after restart, not sure though?

micklynch, 2019-05-20:
Can anyone confirm how do, and what happens if, I uninstall?

martin31821, 2019-05-20:
pstate is resetted after every boot to full frequency range and enabled turbo boost.

digiberk, 2019-09-10:
> Can anyone confirm how do, and what happens if, I uninstall?

@micklynch Simply uninstall the extension from extensions.gnome.org and reboot. I can confirm that upon rebooting, my CPU governor was set back to normal.

fin-ger, 2020-03-31:
This is being worked on in #107

commit 10383d7a (2020-03-30): Add uninstall button, fixes #58
closed 2020-04-05 by fin-ger