deinstapel/cpupower #62: Thermal throttling?

piegamesde, 2018-05-15:
There is this nice script that does thermal throttling: https://github.com/Sepero/temp-throttle/

It's just a loop polling the temperature and changing the max frequency until that temperature is below a threshold.

I think it would be a nice addition to have such features and not that hard to implement. (Since it is using another interface than this extension, it might be faster to re-implement it from scratch)

martin31821, 2018-05-16:
I don't think this is an issue, because the CPU will throttle itself on high temperatures.

piegamesde, 2018-05-16:
Yes, but the only way this is done is through thermald and its configuration possibilities are really limited. This feature would allow thermal throttling at arbitrary temperatures. My Laptop for example throttles at 95°C, which is way too high for me. I'd prefer to setting this down to a temperature where the fan isn't needed (like ~75°C).

victor-bayas, 2020-06-16:
> Yes, but the only way this is done is through thermald and its configuration possibilities are really limited. This feature would allow thermal throttling at arbitrary temperatures. My Laptop for [...]

I know this is an old thread but you can setup thermald config file to throttle your CPU when it reaches 75 °C

piegamesde, 2020-06-16:
@victor-bayas That'd be awesome! I know this isn't the correct thread for this, but do you have any examples or documentation? I still can't find anything about this.

victor-bayas, 2020-06-16:
> @victor-bayas That'd be awesome! I know this isn't the correct thread for this, but do you have any examples or documentation? I still can't find anything about this.

man thermal-conf.xml

In the example 1 change the temperature, by default it's on 86 degrees.

Put that file in etc/thermald