deinstapel/cpupower #64: Gnome-shell warning

piegamesde, 2018-06-21:
When I type `gnome-shell -r`, I get:

```
CPU Power Manager-Message: 09:47:17.770: Inside cpupower.CPUFreqIndicator
Gjs-Message: 09:47:17.958: JS WARNING: [/home/piegames/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 169]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
Gjs-Message: 09:47:17.959: JS WARNING: [/home/piegames/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 173]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
CPU Power Manager-Message: 09:47:17.961: 21 100
Gjs-Message: 09:56:30.259: JS WARNING: [/home/piegames/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 205]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
```

# Hardware information:
- CPU Model: Intel i7 7th Gen
- Linux kernel version 4.17.2-1-ARCH
- Gnome shell version: 3.28.2

The extension seems to be working fine though, it's just an annoying spam in my logs.

bryce-lynch, 2018-09-30:
Same, lots of warnings here:

```
Sep 30 07:49:00 local gnome-shell[2420]: Inside cpupower.CPUFreqIndicator
                                         0 _getMinText() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":258]
                                         2 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":168]
                                         4 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         7 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         11 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         12 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
Sep 30 07:49:00 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 169]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
                                         0 _getMaxText() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":263]
                                         2 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":172]
                                         4 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         7 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         11 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         12 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
Sep 30 07:49:00 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 173]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
                                         0 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":182]
                                         2 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         5 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         9 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         10 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
                                         0 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":184]
                                         2 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         5 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         9 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         10 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
                                         0 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":194]
                                         2 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         5 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         9 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         10 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
Sep 30 07:49:00 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 205]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
                                         0 _sampleFreq/getfreq/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":339]
Sep 30 08:13:46 local gnome-shell[2420]: Inside cpupower.CPUFreqIndicator
Sep 30 08:13:46 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 169]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
Sep 30 08:13:46 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 173]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
[... cut ...]
Sep 30 18:22:51 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 169]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
Sep 30 18:22:51 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 173]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
Sep 30 18:22:51 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 205]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2

```

bryce-lynch, 2018-09-30:
Same, lots of warnings here:

```
Sep 30 07:49:00 local gnome-shell[2420]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a stri>
                                         (Note that array.toString() may have been called implicitly.)
                                         0 _getMinText() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":258]
                                         1 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         2 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":168]
                                         3 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         4 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         5 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         6 _parent() ["resource:///org/gnome/gjs/modules/_legacy.js":39]
                                         7 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         8 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         9 _Base.prototype._construct() ["resource:///org/gnome/gjs/modules/_legacy.js":18]
                                         10 newClass() ["resource:///org/gnome/gjs/modules/_legacy.js":114]
                                         11 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         12 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
                                         13 () ["self-hosted":1080]
Sep 30 07:49:00 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 169]: Too many arguments to metho>
Sep 30 07:49:00 local gnome-shell[2420]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a stri>
                                         (Note that array.toString() may have been called implicitly.)
                                         0 _getMaxText() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":263]
                                         1 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         2 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":172]
                                         3 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         4 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         5 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         6 _parent() ["resource:///org/gnome/gjs/modules/_legacy.js":39]
                                         7 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         8 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         9 _Base.prototype._construct() ["resource:///org/gnome/gjs/modules/_legacy.js":18]
                                         10 newClass() ["resource:///org/gnome/gjs/modules/_legacy.js":114]
                                         11 enable/</<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js":81]
                                         12 spawn_process_check_exit_code/<() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js":50]
                                         13 () ["self-hosted":1080]
Sep 30 07:49:00 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 173]: Too many arguments to metho>
Sep 30 07:49:00 local gnome-shell[2420]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a stri>
                                         (Note that array.toString() may have been called implicitly.)
                                         0 _createMenu() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":182]
                                         1 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         2 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js":75]
                                         3 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
                                         4 _parent() ["resource:///org/gnome/gjs/modules/_legacy.js":39]
                                         5 _init() ["/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js":108]
                                         6 wrapper() ["resource:///org/gnome/gjs/modules/_legacy.js":82]
[... cut ...]
Sep 30 18:22:51 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 169]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
Sep 30 18:22:51 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 173]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2
Sep 30 18:22:51 local gnome-shell[2420]: JS WARNING: [/home/user/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 205]: Too many arguments to method Clutter.Actor.add_child: expected 1, got 2

```

martin31821, 2018-10-01:
That's something I'm already looking into, since Gnome updated their JS engine which fixed several misbehaviors and misinterpretations of the ES spec.

mastercaution, 2020-02-29:
> When I type `gnome-shell -r`, I get:
>
> ```
> CPU Power Manager-Message: 09:47:17.770: Inside cpupower.CPUFreqIndicator
> Gjs-Message: 09:47:17.958: JS WARNING: [...]

This seems to be outdated because the warnings were already fixed in #60 ... looking at the time stamps it was already outdated when the issue was opened.

Instead, I have some new warnings xD
```
[...]
CPU Power Manager-Message: 20:31:00.653: Usage of object.actor is deprecated for PanelMenuButton
get@resource:///org/gnome/shell/ui/environment.js:249:29
CPUFreqBaseIndicator@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:49:9
CPUFreqIndicator@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:57:9
enable/</<@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34
spawn_process_check_exit_code/<@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13

Gjs-Message: 20:31:00.687: JS WARNING: [/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 296]: reference to undefined property "_power_state"
CPU Power Manager-Message: 20:31:00.717: Usage of object.actor is deprecated for Slider
get@resource:///org/gnome/shell/ui/environment.js:249:29
createMenu@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:199:9
CPUFreqIndicator@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:9
enable/</<@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34
spawn_process_check_exit_code/<@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13

CPU Power Manager-Message: 20:31:00.719: Usage of object.actor is deprecated for Slider
get@resource:///org/gnome/shell/ui/environment.js:249:29
createMenu@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:208:9
CPUFreqIndicator@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:9
enable/</<@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34
spawn_process_check_exit_code/<@/home/chrisp/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13
[...]
```
The second warning (undefined property) is my fault but I know how to fix it:
Update L[296](https://github.com/martin31821/cpupower/blob/58b131e3edf143f89bba0be9811dff306d9a2eec/src/indicator.js#L296) from 
`this.powerActions(this._power_state);` to 
`if (this._power_state) this.powerActions(this._power_state);`
ez ;)

### Environment
OS: Ubuntu 19.10
Gnome-Shell: 3.34.1
Linux Kernel: 5.3.0-40

mastercaution, 2020-03-02:
After some experimenting I think I have a fix or the "object.actor is deprecated" warnings

closed 2020-04-14 by fin-ger