deinstapel/cpupower #71: Error: "your computer does not support intel_pstate"

Ricky-Tigg, 2019-02-22:
OS: _Fedora_; Component:  gnome-shell-extension-common  3.30.1

Error message displayed by _Cpupower_: "_your computer does not support intel_pstate_" despite requirements seem met.
```
$ grep -i pstate /boot/config-$(uname -r)
CONFIG_X86_INTEL_PSTATE=y
```
[image]

CPU frequency service is operational as illustrated with the use of _Vitals_ add-on:

[image]

martin31821, 2019-02-22:
Please paste the output of `lscpu` either shortened here or to a paste service.
Having the kconfig set is only one requirement. 
Yout CPU needs to support pstate as well.

Ricky-Tigg, 2019-02-22:
```
$ lscpu | sed -n '1p;13p;$p'
Architecture:        x86_64
Model name:          Intel(R) Core(TM) i5 CPU       M 430  @ 2.27GHz
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 popcnt lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid dtherm ida arat flush_l1d
```

martin31821, 2019-02-22:
As I mentioned, your CPU doesn't support intel_pstate. 
pstate is only supported on sandy bridge and newer CPUs, so there's nothing I can do here.

closed 2019-02-22 by martin31821

Ricky-Tigg, 2019-02-22:
You must have able to determine that state from the flags listed in the output. Then it must be so. Bye.

DevNM21, 2020-05-29:
Hey guys I'm facing the same issue, my OS is pop!_os 20, here's my output of lscpu:
```Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   43 bits physical, 48 bits virtual
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              2
Core(s) per socket:              2
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       AuthenticAMD
CPU family:                      23
Model:                           17
Model name:                      AMD Ryzen 3 2200U with Radeon Vega Mobile Gfx
Stepping:                        0
Frequency boost:                 enabled
CPU MHz:                         1586.795
CPU max MHz:                     2500.0000
CPU min MHz:                     1600.0000
BogoMIPS:                        4991.14
Virtualization:                  AMD-V
L1d cache:                       64 KiB
L1i cache:                       128 KiB
L2 cache:                        1 MiB
L3 cache:                        4 MiB
NUMA node0 CPU(s):               0-3
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Full AMD retpoline, IBPB conditional, STIBP disabled, RSB fil
                                 ling
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 c
                                 lflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm 
                                 constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni p
                                 clmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx 
                                 f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misaligns
                                 se 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpex
                                 t perfctr_llc mwaitx cpb hw_pstate sme ssbd sev ibpb vmmcall fsgsbase bmi
                                 1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv
                                 1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_sc
                                 ale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_v
                                 msave_vmload vgif overflow_recov succor smca
```
there's hw_pstate, is that any helpful?

fin-ger, 2020-05-29:
No, unfortunately this extension **only** supports **Intel** CPUs, meaning it **must** support **intel_pstate**.

However, you are free to contribute support for other CPU vendors. We just don't have regular access to e.g. AMD CPUs and therefore do not maintain support for them.