deinstapel/cpupower #72: Request: port the extension to Linux Mint Cinnamon

dannyk81, 2019-03-03:
This extension is great! as a Linux Mint user I really miss it, would you consider porting it to Cinnamon?

fin-ger, 2019-03-04:
As neither @martin31821 or me is using Cinnamon on a daily basis, we cannot provide the same stability and quality as we are providing for Gnome. Also we both have no understanding on how extensions for Cinnamon are written. Can we even share code between this and a port? Anyway we would need a maintainer who can provide a stable and well integrated experience with Cinnamon.

Anybody willing to do this?

martin31821, 2019-03-05:
I looked a bit into the cinnamon extension structure and it's very similar to the gnome one, so the only thing we might need to change is the way the UI works. 
I'm unsure about the prefs dialog though. 

Otherwise I'd also like to see a third party maintaining this port since I don't use cinnamon.

MorenoMdz, 2019-07-01:
I came here to ask for this and saw this issue, so yeah, +1 for Cinnamon.

fin-ger, 2020-04-14:
Is anyone interested in developing a Cinnamon port?

fin-ger, 2020-04-21:
Please reopen if anyone is interested in porting this extension

closed 2020-04-21 by fin-ger