deinstapel/cpupower #73: Total system freeze when running World of Warcraft via Lutris

ebeyer, 2019-03-13:
I installed the extension through firefox at extensions.gnome.org.  I set speed to maximum.  While playing WoW, about 5 minutes in, all the video on all monitors (3 on this system) completely freeze up.  I can still move the mouse but nothing responds to clicks.  No response to any keyboard commands or sequences, requiring a manual reset.  Removing the extension eliminates the error.

# Creating a new issue

* Check if another issue with the same problem exists
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
  * policykit working and rule working
* Check if the cpufreqctl utility works correctly
* Please include the following information (only for bugreports):
  * CPU Model
  * Linux kernel version
  * policykit version
  * Gnome shell version
  * journalctl -xb (maybe pastebin/filtered)

martin31821, 2019-03-13:
Please fill in the issue template (esp. CPU model, and your versions) and also install a temperature monitoring tool to ensure your problem is not a thermal one.

To further chase it down: Can you reproduce this behavior even if you don't set anything within the extension?

fin-ger, 2020-04-14:
Is this still an issue?

fin-ger, 2020-04-21:
Please reopen if this is still an issue

closed 2020-04-21 by fin-ger