deinstapel/cpupower #74: Broken on Gnome 3.32

Sangeppato, 2019-03-16:
Currently the extension doesn't appear in the top panel in Gnome 3.32, other extensions have similar problems so it could be about some Gnome changes.

martin31821, 2019-03-16:
I'm not yet on 3.32 but I'm going to look into it as soon as I can update.

martin31821, 2019-03-16:
Started to work on it: https://github.com/martin31821/cpupower/tree/fix/3.32
Seems like a bigger change...

martin31821, 2019-03-17:
If you like, you can check out the 3.32 branch and do some testing.
I think, we also might need to change the minimum supported versions from 3.10 to 3.30ish, because there are big changesdue to the introduction of ES6 classes.

K5HV, 2019-03-17:
fix/3.32 seems to be working with 3.32, at least for me.
Thank you!

Sangeppato, 2019-03-17:
Working here as well, thank you very much!
I've just noticed two small issues:

* The dot to indicate the currently used profile is not displayed
* If I set a very low minimum frequency, the slider is not reset to the actual hardware minimum frequency as it used to do (It can even stay at 0%)

martin31821, 2019-03-18:
> * The dot to indicate the currently used profile is not displayed

Fixed in 4da44c2

> * If I set a very low minimum frequency, the slider is not reset to the actual hardware minimum frequency as it used to do (It can even stay at 0%)

That's kernel dependent. It seems that with 5.0 the frequency is not reset anymore.

martin31821, 2019-03-22:
Update is available on extensions.gnome.org, so fixed now.

closed 2019-03-22 by martin31821