deinstapel/cpupower #78: Stops working after system wakes up from sleep

iggy12345, 2019-05-07:
# Creating a new issue

* Check if another issue with the same problem exists
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
  * policykit working and rule working
* Check if the cpufreqctl utility works correctly
* Please include the following information (only for bugreports):
  * CPU Model: intel i5 3230m
  * Linux kernel version: ubuntu 18.04
  * policykit version: I don't know, your README says it ships default
  * Gnome shell version: 18.04?
  * journalctl -xb (maybe pastebin/filtered)
https://pastebin.com/AtZJ47GJ

The extension works after a full shutdown and restart, but if I put my computer to sleep (close the lid) then it stops working, it still displays the frequency of the CPU correctly, but none of the presets are able to control it (it runs as fast or as slow as it wants regardless of the limits put in place)

[image]

martin31821, 2019-05-10:
That seems a bit odd to me, but I don't have the exact CPU model to reproduce this here...

Are the values for minimum and maximum frequency jumping around or does the CPU simply not respect frequency limits?

iggy12345, 2019-05-14:
The CPU simply does not respect the frequency limits

martin31821, 2019-05-17:
If the cpu doesn't respect the set limits, it's unfortunately nothing I can fix.

fin-ger, 2020-04-14:
If a fix for this issue is known, please share it

closed 2020-04-14 by fin-ger