deinstapel/cpupower #79: Icon color doesn't follow GNOME Shell theme color

maxjf1, 2019-05-23:
I'm using an custom GNOME shell theme (MacBuntu) that use dark icons and text on the shell.

The extension doesn't follow the theme color and uses an gray icon (other gnome extensions are using custom icons and the color is alright).

[image]

[image]

guekho64, 2019-06-28:
I'm not completely aware of the actual behaviour of this program, nor how it works internally, but, In my opinion, It'd be better to add one of these:

* An option to "Turn ON/OFF dark theme" (Only switching the icon from white to black should suffice...)
* Or implementing a way to let users add a custom icon to be displayed instead of the original one

Both would solve this specific issue, so I have no preference over one them.

If any of these options gets implemented, I can provide the dark icon if needed (It's not like it's that hard anyways...)

maxjf1, 2019-07-01:
> I'm not completely aware of the actual behaviour of this program, nor how it works internally, but, In my opinion, It'd be better to add one of these:
>
> * An option to "Turn ON/OFF dark theme" (Only [...]

i think the problem isn't an dark and light icon

Gnome automatically set monochrome icon colors if they follow some pattern (wich i dont know)

For instance, i am now using an dark theme and it makes gnome icons completly white. this plugin uses an gray icon no matter the gnome settings. 

I don't know what is needed to make gnome automatically set the icon color, maybe using SVG or native icon...

[image]

commit 10b35a1b (2019-09-24): Add cpu-symbolic icon

* Fixes blurry icon
* Follows GTK theme color. Fixes #79
closed 2019-09-24 by martin31821