deinstapel/cpupower #82: Can't go from low profile to high anymore

Anon-Exploiter, 2019-07-01:
# Can't go from low profile to high anymore

* System Information
  * Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
  * Linux ... 4.18.0-20-generic #21~18.04.1pop0-Ubuntu SMP Wed May 15 14:41:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
  * pkaction version 0.105
  * GNOME Shell 3.28.4
  * https://pastebin.com/raw/zCLyxkeT

Problem: Sometimes, I add downloading of stuff/torrents on my laptop and change the profile to Energy Saver mode and it shuts down automatically when the battery runs out. Today, when I tried changing the profile to High It just isn't changing. The usage is still to 0.80 GHz when on high it went to 3.50 GHz. Even though, now the profile is set to high. The processes get stuck while running, the lag is obvious. I tried uninstalling & installing the package again, that didn't help at all. 

[image]

Basically what, I'd like to ask is how I can again change the state to high again. Thanks much!

Anon-Exploiter, 2019-07-06:
Uninstalled, that fixed it :)

closed 2019-07-06 by Anon-Exploiter

fin-ger, 2019-07-06:
Hmm ok, this shouldn't happen. But yes, uninstalling will prevent the extension from restoring any previous configuration. The kernel always uses the system defaults on boot.