deinstapel/cpupower #83: installer.sh doesn't work on ostree (Silverblue)

krashekspres, 2019-08-07:
Installer.sh fails on Fedora Silverblue:
```
 ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/installer.sh  install
Installing policykit action... /home/drumpf/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/installer.sh: line 56: /usr/share/polkit-1/actions/mko.cpupower.setcpufreq.policy: Read-only file system
Failed
```
* Please include the following information (only for bugreports):
  * CPU Model: `i7 5500u`
  * Linux kernel version: `5.2.5-200.fc30.x86_64`
  * Gnome shell version: `3.32`
  * OS: Fedora Silverblue 30

serverwentdown, 2020-03-25:
I believe there is no solution to running cpupower in Silverblue, because it currently relies on the existence of a shell script and a PolicyKit action. Creating a PolicyKit action will require a package to be installed, but that makes this extension dependent on the distribution.

fin-ger, 2020-03-29:
I will add instructions for package maintainers to the README when #95 is done. You can then package and install this extension via the package manager of your distribution.

fin-ger, 2020-04-05:
#107 got merged into master, which adds instructions to the README for package maintainers.

fin-ger, 2020-04-14:
As v9.0.0 got released, packaging for Silverblue could start. As am not using Fedora Silverblue, nor have I any experience packaging for Silverblue, I am asking for help here. Anyone interested in packaging?

fin-ger, 2020-04-28:
As Fedora Silverblue is focused on Flatpak and the Flatpak developers noted that [supporting installation of Gnome Shell Extensions is out-of-scope](https://github.com/flatpak/flatpak/issues/481), I will label this issue as `wontfix` until a solution pops up for installing Gnome Shell Extensions via Flatpak.

Please reopen if you know of such a solution.

closed 2020-04-28 by fin-ger