deinstapel/cpupower #84: Only works for 1 user at any time

georgekaz, 2019-09-01:
Using the extension on an 18.04 laptop with 2 different users (1 with sudo, the other without) , it works for 1 user at a time. It says "Installation required; Attempt installation". I install (prompted for sudo creds) and the extension works. Switch to the other user and the extension has stopped and reads "Installation required; Attempt installation". Do the same and it works, switch back to the first user and it's disabled again. i.e. it works, but only for 1 user at a time.

* Please include the following information (only for bugreports):
  * CPU Model - Intel(R) Core(TM) i7-5600U CPU @ 2.60GHz
  * Linux kernel version - Linux goldfinger 5.0.0-25-generic #26~18.04.1-Ubuntu SMP Thu Aug 1 13:51:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
  * policykit version - policykit-1 (0.105-20ubuntu0.18.04.5).
  * Gnome shell version - GNOME Shell 3.28.4
  * journalctl -xb (maybe pastebin/filtered)

P1zz4br0etch3n, 2019-10-04:
Same for me on Fedora 30.

fin-ger, 2019-10-04:
I think this is due to the `CFC` variable pointing to a specific user installation when creating the polkit action.

https://github.com/martin31821/cpupower/blob/master/src/installer.sh#L27

You may try installing the extension system wide. However, I don't think this extension is packaged for Fedora. Can anyone try manually placing the extension files into `/usr/share/gnome-shell/extensions/` and see if it works?

> :bangbang: Doing this may break your system. Do this only if you know what you are doing! :bangbang:

---

Another idea would be to generate polkit actions per user. Currently, the polkit action of all users are stored in the same place (`/usr/share/polkit-1/actions/mko.cpupower.setcpufreq.policy`).

As neither @martin31821 nor me are using this extension in a multi-user setup, we would need someone who tests the multi-user code prior to a release to prevent regression bugs. Anyone interested?

P1zz4br0etch3n, 2019-10-04:
I moved the extension folder from .local/share/gnome-shell/extensions to /usr/share/gnome-shell/extensions. But after that it doesn't appear in the extensions list in neither gnome-software nore gnome-tweaks.

Will try the other idea later if nobody else tries it first.

cykl, 2020-01-27:
> As neither @martin31821 nor me are using this extension in a multi-user setup, we would need someone who tests the multi-user code prior to a release to prevent regression bugs. Anyone interested?

@fin-ger Fedora user here, also volunteer to test multi-user support. I confirm that cpupower extension is not in Fedora repos. Would a system wide package solve the issue?

I do confirm the issue is that rule content contains a user specific path but rule path is shared between all users. Adding user name or UID to the rule path would likely solve the issue. Haven't thought about backward compatibility yet.

georgekaz, 2020-01-27:
@fin-ger I'm happy to test on Ubuntu 18.04.

fin-ger, 2020-01-27:
This sounds great! I think we should support both multi-user scenarios:

1. Installing the extension via extensions.gnome.org per user and provide per-user policies
2. Installing the extension system-wide for all users

However, I'm currently busy with exams, I'll see if I can look into it. Otherwise, this comes on my todo list for March.

mastercaution, 2020-03-28:
> 1. Installing the extension via extensions.gnome.org per user and provide per-user policies

Do we need this per-user-install support? 
Due to #102 `cpufreqctl` need to be installed in a root directory. The policykit action can point to the new location but this will give all users access to it. All programs installed with a package manager are also installed for all users, so it's nothing new.
However, if someone doesn't want all users to be able to execute `cpufreqctl` they can simply take away the execute-permission for that user in the same way they would do with other programs installed by a package manager (Unix file permissions).

Otherwise, we have to fiddle with [Polkit authorization rules](https://wiki.archlinux.org/index.php/Polkit#Authorization_rules) and adding a rule for every user who installs it and removing it when he uninstalls it. It's possible but I don't know if this is needed.

fin-ger, 2020-03-31:
This is being worked on in #107

closed 2020-04-14 by fin-ger