deinstapel/cpupower #87: panel bar icon location

Ace-Cassidy, 2019-09-29:
Issue: icon for cpupower is to the far left, but I want clipboard manager on far left

bug; cpu power does not set status area position priority at all!

potential solution:
in extension.js
Main.panel.addToStatusArea('cpupower', _indicator._mainButton); => Main.panel.addToStatusArea('cpupower', _indicator._mainButton, 2);

fin-ger, 2019-10-01:
Hmm, I don't know if there are any upstream conventions by the gnome developers how the icons should appear in the top bar. Maybe we can make this configurable...

Does anyone know how this is handled in other extensions or by upstream gnome-shell?

/cc @martin31821

JustCryen, 2021-11-14:
My extension appeared on the right side of the panel and I wanted it on the left at the end of other extensions.
So I changed this line 
```
function enableIndicator(instance) {
    Main.panel.addToStatusArea("cpupower", instance.mainButton);
    instance.enable();
}
```
To this 
```
function enableIndicator(instance) {
    Main.panel._addToPanelBox("cpupower", instance.mainButton, 4, Main.panel._leftBox);
    instance.enable();
}

```
In my instance `4` sets the lowest priority (so it appears at the end of the list) and `Main.panel._leftBox` makes it switch to the left side of the panel.
It would be nice to have it configurable from the extension menu though.

JustCryen, 2022-04-28:
Can we somehow move forward with this?
I'm kinda tired of changing this manually every release ;/
I'm also not really familiar with how to implement a gui setting for this so I can't really create a pull request, sorry.

JustCryen, 2022-07-04:
Hi
I'm trying to implement this feature myself to make a pull request later.
As stated previously, I have a working solution but I would like to add a gui entry to make it permanent and easily selectable.
Thus I need a little bit more help on how to do this.

I've never tried javascript before but I managed to add this gui entry and I just need to "link" it to a specific place in the script.

Diff for `extension.js`
```diff
52c52
<     Main.panel.addToStatusArea("cpupower", instance.mainButton);
---
>     Main.panel._addToPanelBox("cpupower", instance.mainButton, 5, Main.panel._leftBox);
```
This makes the extension appear on the left side of the panel, thanks to the lower priority it's also added last to that side making it appear on the right.
[image]

Diff for `cpupower-preferences.glade`
```diff
303a304,381
>                           <object class="GtkLabel" id="PositionInPanelAsLabel">
>                             <property name="visible">True</property>
>                             <property name="can-focus">False</property>
>                             <property name="halign">start</property>
>                             <property name="hexpand">True</property>
>                             <property name="label" translatable="yes">Set position to</property>
>                             <accessibility>
>                               <relation type="label-for" target="PositionInPanelComboBox"/>
>                             </accessibility>
>                           </object>
>                           <packing>
>                             <property name="left-attach">0</property>
>                             <property name="top-attach">10</property>
>                           </packing>
>                         </child>
>                         <child>
>                           <object class="GtkComboBoxText" id="PositionInPanelComboBox">
>                             <property name="visible">True</property>
>                             <property name="can-focus">False</property>
>                             <property name="halign">start</property>
>                             <property name="hexpand">False</property>
>                             <property name="tooltip-text" translatable="yes">Set the position of the extension on the panel</property>
>                             <items>
>                               <item id="pos_left" translatable="yes">Left</item>
>                               <item id="pos_center" translatable="yes">Center</item>
>                               <item id="pos_right" translatable="yes">Right</item>
>                             </items>
>                             <signal name="notify::active" handler="onPositionInPanelComboBoxActiveNotify" swapped="no"/>
>                             <accessibility>
>                               <relation type="labelled-by" target="PositionInPanelAsLabel"/>
>                             </accessibility>
>                             <child internal-child="accessible">
>                               <object class="AtkObject" id="PositionInPanelComboBox-atkobject">
>                                 <property name="AtkObject::accessible-name" translatable="yes">Position in Panel</property>
>                                 <property name="AtkObject::accessible-description" translatable="yes">Set the position of the extension on the panel</property>
>                               </object>
>                             </child>
>                           </object>
>                           <packing>
>                             <property name="left-attach">1</property>
>                             <property name="top-attach">10</property>
>                           </packing>
>                         </child>
>                         <child>
>                           <object class="GtkComboBoxText" id="PriorityInPanelComboBox">
>                             <property name="visible">True</property>
>                             <property name="can-focus">False</property>
>                             <property name="halign">end</property>
>                             <property name="hexpand">False</property>
>                             <property name="tooltip-text" translatable="yes">Set the priority of the extension on the panel</property>
[... 10 lines cut ...]
>                             </items>
>                             <signal name="notify::active" handler="onPositionInPanelComboBoxActiveNotify" swapped="no"/>
>                             <accessibility>
>                               <relation type="labelled-by" target="PositionInPanelAsLabel"/>
>                             </accessibility>
>                             <child internal-child="accessible">
>                               <object class="AtkObject" id="PriorityInPanelComboBox-atkobject">
>                                 <property name="AtkObject::accessible-name" translatable="yes">Priority in Panel</property>
>                                 <property name="AtkObject::accessible-description" translatable="yes">Set the priority of the extension on the panel</property>
>                               </object>
>                             </child>
>                           </object>
>                           <packing>
>                             <property name="left-attach">1</property>
>                             <property name="top-attach">10</property>
>                           </packing>
>                         </child>
>                         <child>
```

I based it heavily on the "Show frequency" box.
For now it looks something like this, obviously controls are not yet linked.
I also couldn't figure out how to add the spacing between the new entry and a previous one.
How would I continue from that point on?
[image]