deinstapel/cpupower #95: Unattended installation

valpo, 2019-12-18:
I would like to distribute this extension to our all our clients. The installation can be easily done by script, but the extension still asks the user for admin rights for the policy kit rule. Our users do not have admin rights. 
I am trying to trigger the installation from the install script (running as root) with this line: 
`/home/AD/$user/.local/share/gnome-shell/extensions/$EID/src/installer.sh install`
Is there a way to install the policy kit rule from a script?

fin-ger, 2019-12-19:
The policy kit rule is installed with the `installer.sh install` command.

Please post the content of `/usr/share/polkit-1/actions/mko.cpupower.setcpufreq.policy` here after running the install command.

@martin31821 we are still using `installer.sh check` to check for correct installation, right?

martin31821, 2019-12-19:
installer.sh install should be enough to have it. The only possible cause I can think of right now, would be that some script messes up the binary path. 

xref: #34

valpo, 2019-12-19:
Looks like you are right, I messed it up somehow. Maybe the test-machine is different, it works on other machines. Just in case anyone is interested in, we are using a script like this for the distribution:
```bash
apt-get -qy install cpufrequtils
cd /tmp # here we have download the extension from an internal server
EID=$(unzip -c cpupowermko-sl.de.v17.shell-extension.zip metadata.json | grep uuid | cut -d \" -f4)
cd /home/AD # here we see which users are logged into the machine
for user in *; do
	su $user << EOF
  mkdir -p /home/AD/$user/.local/share/gnome-shell/extensions/$EID
  unzip -q /tmp/cpupowermko-sl.de.v17.shell-extension.zip -d /home/AD/$user/.local/share/gnome-shell/extensions/$EID/
  gnome-shell-extension-tool -e $EID
EOF
  # install policy kit rule
  /home/AD/$user/.local/share/gnome-shell/extensions/$EID/src/installer.sh install
done
```
I assume the last line could run after the loop?

martin31821, 2019-12-27:
@valpo you would need to rename the generated cpupower policy for each user at the moment. In this case we should implement a fallback or a possibility to have a custom cpufreqctl location and custom generated policy.

valpo, 2020-01-05:
@martin31821  thanks for the hint. It is my first contact with policykit, but now I understand the problem (and btw I know what went wrong during my tests).

serverwentdown, 2020-03-25:
I think we should change the installer to install `cpufreqctl` into `/usr/local/bin/`. See issue #102

fin-ger, 2020-03-31:
This is being worked on in #107

closed 2020-04-14 by fin-ger