deinstapel/cpupower #97: Missing UI updates

mastercaution, 2020-02-27:
During my tests on the auto-switch feature a few days ago I discovered a little issue in `indicator.js`.

The `Current Frequency:` value is not updating on the indicator popup UI. 
And also the two sliders and the boost switch states are only updated when you open the preferences. You can see this if you set the minimum frequency to 0%, then the pstate driver should instantly set the minimum frequency back to the hardware minimum (mostly not 0) but the slider stays at 0% until you open up the preference menu.

Is this behavior wanted? 

I looked in the code and it doesn't seem wanted, because the values are updated but not displayed.
At line [361](https://github.com/martin31821/cpupower/blob/8e26d4a46abb0ed3b4207c9e6133d8c3ca716157/src/indicator.js#L361) and [368](https://github.com/martin31821/cpupower/blob/8e26d4a46abb0ed3b4207c9e6133d8c3ca716157/src/indicator.js#L368) the `!this.menu.isOpen` causes the issue. It should be `this.menu.isOpen` to update the UI and work as expected.

fin-ger, 2020-02-27:
This sounds like a bug. Could you create a PR so we can fix this?

mastercaution, 2020-02-27:
Of course ;)

PR #98 (2020-02-29): Fixed missing UI updates
This fixes #97
closed 2020-02-29 by martin31821