deinstapel/cpupower #99: Extension icon dispears from latest Ubuntu 20.04 update

StephanieSunshine, 2020-03-02:
The latest Ubuntu 20.04 Gnome update seems to have broke a handful of extensions.  After trying numerous times to start, top, and reinstall this extension, both Gnome Extensions from the browser and from Gnome Tweaks would say that the extension was running without issue, however, with out an icon showing or anything clickable in the location where it used to be.  I pulled your git repo and tried to run your installer by hand, which worked once I gave it sudo.  Your app came up this time and said it wanted to install something, it gave me a prompt, I gave it a password and I'm back to a broken state again.  I happen to find this in `/var/log/syslog` that I think is a hint to what is going on, but I'm not familiar enough with this project or Javascript to just debug this.

```
Mar  1 16:17:26 laputa gnome-shell[8936]: Supported
Mar  1 16:17:26 laputa gnome-shell[8940]: Installed
Mar  1 16:17:26 laputa gnome-shell[7098]: Usage of object.actor is deprecated for PanelMenuButton#012get@resource:///org/gnome/shell/ui/environment.js:249:29#012CPUFreqBaseIndicator@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:49:9#012CPUFreqIndicator@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:57:9#012enable/</<@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34#012spawn_process_check_exit_code/<@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13
Mar  1 16:17:26 laputa gnome-shell[7098]: JS ERROR: Error: Tried to construct an object without a GType; are you using GObject.registerClass() when inheriting from a GObject type?#012CPUFreqProfileButton@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profilebutton.js:37:9#012createMenu@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:167:33#012CPUFreqIndicator@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:9#012enable/</<@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34#012spawn_process_check_exit_code/<@/home/stephanie/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13
Mar  1 16:17:26 laputa systemd[6793]: Started Application launched by gnome-shell.
Mar  1 16:17:26 laputa systemd[6793]: message repeated 2 times: [ Started Application launched by gnome-shell.]
```

Let me know if there is anything more I can provide.  
Linux kernel 5.5.6, Ubuntu 20.04, Gnome 3.34.2

martin31821, 2020-03-02:
cc @fin_ger

mastercaution, 2020-03-09:
I have a fix for that in my new branch https://github.com/mastercaution/cpupower/tree/gnome-3.36-support. It is a work-in-progress.

Due to a [bug](https://gitlab.gnome.org/GNOME/gnome-shell/issues/2317) in Gnome Shell 3.35.91 (currently the latest release for Ubuntu 20.04), which was possibly fixed in 3.35.92, it is not possible to open the preferences window.

martin31821, 2020-03-10:
@mastercaution I just tested your 3.36 branch, I like the idea using parseFloat() :smile:

brneor, 2020-03-20:
> I have a fix for that in my new branch https://github.com/mastercaution/cpupower/tree/gnome-3.36-support. It is a work-in-progress.
>
> Due to a [...]

Sorry for being a total noob but how do I install the extension from your branch? :)

mastercaution, 2020-03-20:
> Sorry for being a total noob but how do I install the extension from your branch? :)

No problem ;)
1. Download the source code of the branch as zip
2. Unzip it and rename the folder from `cpupower-gnome-3.36-support` to `cpupower@mko-sl.de`
3. Move the folder to ~/.local/share/gnome-shell/extensions/
4. Restart Gnome Shell with `[ALT+F2] -> r -> [Enter]`

brneor, 2020-03-20:
@mastercaution thank you, you're an angel! hehe 

I literally can't use my laptop on battery without this extension.

martin31821, 2020-03-26:
fixed in #104

closed 2020-03-26 by martin31821