deinstapel/cpupower #101: Does not show icon after upgrade to Gnome shell 3.36

manhtukhang, 2020-03-09:
## System details:
  * OS: `Archlinux`
  * Kernel: `5.5.8`
  * Gnome shell version: `3.36.0`
## Bug:
After upgraded Gnome shell from `3.34` to `3.36` over Archlinux Pacman auto upgrading, app still working but top pannel icon does not show
## Error logs:
```js
Mar 09 15:41:58 MT gnome-shell[12030]: Usage of object.actor is deprecated for PanelMenuButton
                                       get@resource:///org/gnome/shell/ui/environment.js:285:29
                                       CPUFreqBaseIndicator@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:49:9
                                       CPUFreqIndicator@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:57:9
                                       enable/</<@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34
                                       spawn_process_check_exit_code/<@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13
Mar 09 15:41:58 MT gnome-shell[12030]: JS ERROR: Error: Tried to construct an object without a GType; are you using GObject.registerClass() when inheriting from a GObject type?
                                       CPUFreqProfileButton@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profilebutton.js:37:9
                                       createMenu@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:167:33
                                       CPUFreqIndicator@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:14
                                       enable/</<@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34
                                       spawn_process_check_exit_code/<@/home/tukm/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13
```

mastercaution, 2020-03-09:
This is the same error as in #99.

martin31821, 2020-03-09:
duplicate of #99

closed 2020-03-09 by martin31821