deinstapel/cpupower #102: Unsafe Installation

serverwentdown, 2020-03-25:
**This is a really important security issue**. I think we should change the installer to install `cpufreqctl` into `/usr/local/bin/`. 

Allowing users to run scripts installed in the user folder is completely unsafe, effectively providing root privileges to any program running as the current user.

Example exploit, run as the user:
```
mv $HOME/.local/share/gnome-shell/extensions/cpufreqctl $HOME/old # Succeeds because the directory is user writeable
echo "touch /root/pwned" > $HOME/.local/share/gnome-shell/extensions/cpufreqctl
chmod +x $HOME/.local/share/gnome-shell/extensions/cpufreqctl
```

Instead, the installer should drop `cpufreqctl` into `/usr/local/bin` or rely on a system program. 

These [lines](https://github.com/martin31821/cpupower/blob/58b131e3edf143f89bba0be9811dff306d9a2eec/src/installer.sh#L59-L62) attempted to fix the issue, but directories that are writeable can be renamed. 

This also happens to fix #95 :smile:

mastercaution, 2020-03-27:
Very good that you spotted it, it's very important that we fix this! 

Moving `cpufreqctl` to `/usr/local/bin` is a good idea in my eyes, but it means, that `cpufreqctl` will be installed for all users.
So we also have to work on #84 for customizable multi-user support ... if this is needed.

fin-ger, 2020-03-29:
The current idea is to add a Makefile to the project with the following targets:

```
install # installs extension to $PREFIX/share/gnome-shell/extensions
uninstall # uninstalls extension from $PREFIX/share/gnome-shell/extensions
install-tool # installs cpufreqctl to $PREFIX/bin and polkit
uninstall-tool # uninstalls cpufreqctl from $PREFIX/bin and polkit
```

## extensions.gnome.org workflow

Install it via extensions.gnome.org, where the extension will install the `cpufreqctl` tool to `/usr/loca/bin/cpufreqctl-$userid` and sets up a polkit rule. This is necessary as different users might use different versions of `cpupower`.

## Admins remote install per user workflow

Login to the remote machine and run `make install install-tool PREFIX=/home/worker/.local TOOL_SUFFIX=worker` will install the extension to `/home/worker/.local/share/gnome-shell/extensions/` and the `cpufreqctl` tool to `/usr/local/bin/cpufreqctl-worker` and sets up a polkit rule for `/usr/local/bin/cpufreqctl-worker` for user `worker`.

## Admins remote install global workflow

Login to the remote machine and run `make install install-tool PREFIX=/usr` will install the extension to `/usr/share/gnome-shell/extensions` and the `cpufreqctl` tool to `/usr/bin/cpufreqctl` and sets up a polkit rule for `/usr/bin/cpufreqctl` for normal users (maybe for the wheel group?).

## Package maintainers workflow

Same as admins remote install global workflow.

## Developer workflow (Option 1)

Clone the git, make your changes, and run `make install PREFIX=$HOME/.local`

## Developer workflow (Option 2)

Clone the git directly into `~/.local/share/gnome-shell/extensions`.

# Why a separate `install` and `install-tool` target?

`install` just copies the contents of this repository to the install destination folder, e.g. it behaves exactly like the install button on extensions.gnome.org. This is helpful in maintaining a smooth experience when publishing a new version on extensions.gnome.org, as we (the developers) can test the install with `make install`.

# Problems

I need time to implement this :see_no_evil:

fin-ger, 2020-03-29:
I started working on this

fin-ger, 2020-03-31:
Look at #107 for progress

closed 2020-04-14 by fin-ger

hackel, 2020-10-06:
This tool should not be touching *anything* outside of `/usr/local` and `/etc`.  It would be ideal if you had a simple way to just generate the files so they can be expected and then copied into place manually.  People should *not* be encouraged to grant root access to random GNOME Shell extension (no offence).

fin-ger, 2020-10-07:
If I get this right, you are missing a `make package-tool` target which creates a generated tgz for the polkit rules and the cpufreqctl tool. This can then be manually extracted onto the root filesystem.

The core problem of this is that one generated polkit rule only works for a specific installation path of the extension. If the extension got installed system wide or by another user, you need a different polkit rule as when the extension got installed via the gnome extensions website. This is to prevent file collisions on multi user systems where each user installs the extension individually and to prevent privilege escalations from other users on the same system.

However, we could provide a complete extractable tgz for system wide installations only, in the Github releases download section. This tgz contains the shell extension and the corresponding polkit rules for a system wide installation.

Also, we can provide a `make package-tool TOOL_SUFFIX=... PREFIX=...` copy&paste command as an alternative to the "ask-for-sudo-password-installation". However, I would prefer to have this as an expert option which is not the default, to not make the installation more inaccessible to users.

I would prefer distros to ship this extension. Maybe I'll find the time to create Gentoo and Fedora packages... But I hoped others would get on this as we are mostly lacking time to package for all the distros...