deinstapel/cpupower #106: Undefined property _power_state

amivaleo, 2020-03-29:
Hi!

Not sure if this might be of any interested. I gave `journalctl /usr/bin/gnome-shell` so to check for issues on another gnome extension I use when I noticed several lines like the following.

```
mar 28 09:10:59 jimmy-xps-9360 gnome-shell[1685]: JS WARNING: [/home/jimmy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js 296]: reference to undefined property "_power_state"
                                                  createMenu@/home/jimmy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:199:9
                                                  CPUFreqIndicator@/home/jimmy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:9
                                                  createMenu@/home/jimmy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:208:9
                                                  CPUFreqIndicator@/home/jimmy/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:9
```

I haven't experienced any issue so far, and yet I thought it could be of some help to tell you about this "undefined property _power_state".

mastercaution, 2020-03-29:
This should be fixed in #104 in [this](https://github.com/martin31821/cpupower/pull/104/commits/b0445b1d31c68476370dd3a3b0a56ad0105a7e42) commit. 
Have you updated to the latest version from the git repository?

amivaleo, 2020-03-29:
Well... No. I only update extensions from the website. 😶

fin-ger, 2020-03-29:
This will be fixed with the next update. I will leave this open until the new version is published on extensions.gnome.org

closed 2020-04-14 by fin-ger