deinstapel/cpupower #108: Auto switch dont fully work with auto switch

Tikilou, 2020-03-31:
Hi guys, thanks for this great extension ! I"ve upgrade to Gnome 3.36, and installed the latest revision.
Actually i have specified in settings to disable turbo boost when i unplug AC Adapter, then re-enable it when i plug the Ac Adapter on my laptop.
Well, it work when i unplug the adapter, but when i plug it again, turbo boost is not enabled by the extension.

  * CPU Model : Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz (Turbo boost 3,1Ghz)
  * Linux kernel version : Linux dell-xps-13-9360 5.6.0-1-MANJARO #1 SMP PREEMPT Mon Mar 23 14:46:38 UTC 2020 x86_64 GNU/Linux
  * policykit version : 0.116-4.1
  * Gnome shell version : 3.36.0
  * journalctl -xb : 

`mars 28 13:45:37 dell-xps-13-9360 kernel: microcode: microcode updated early to revision 0xca, date = 2019-09-26
mars 28 13:45:37 dell-xps-13-9360 kernel: Linux version 5.6.0-1-MANJARO (builder@2c68a53ad7e1) (gcc version 9.3.0 (Arch Linux 9.3.0-1)) #1 SMP PREEMPT Mon Ma>
mars 28 13:45:37 dell-xps-13-9360 kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-5.6-x86_64 root=UUID=e6d64d2a-7222-47e4-972f-60153d08510c rw enable_rc6=1 en>
mars 28 13:45:37 dell-xps-13-9360 kernel: KERNEL supported cpus:
mars 28 13:45:37 dell-xps-13-9360 kernel:   Intel GenuineIntel
mars 28 13:45:37 dell-xps-13-9360 kernel:   AMD AuthenticAMD
mars 28 13:45:37 dell-xps-13-9360 kernel:   Hygon HygonGenuine
mars 28 13:45:37 dell-xps-13-9360 kernel:   Centaur CentaurHauls
mars 28 13:45:37 dell-xps-13-9360 kernel:   zhaoxin   Shanghai  
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
mars 28 13:45:37 dell-xps-13-9360 kernel: x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-provided physical RAM map:
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x0000000000059000-0x000000000009efff] usable
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x0000000000100000-0x000000006a61bfff] usable
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000006a61c000-0x000000006a61cfff] ACPI NVS
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000006a61d000-0x000000006a61dfff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000006a61e000-0x00000000701edfff] usable
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000701ee000-0x0000000070578fff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x0000000070579000-0x00000000705bcfff] ACPI data
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000705bd000-0x000000007ac2efff] ACPI NVS
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000007ac2f000-0x000000007b51dfff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000007b51e000-0x000000007b5fefff] type 20
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000007b5ff000-0x000000007b5fffff] usable
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x000000007b600000-0x000000007f7fffff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
mars 28 13:45:37 dell-xps-13-9360 kernel: BIOS-e820: [mem 0x0000000100000000-0x000000027e7fffff] usable
mars 28 13:45:37 dell-xps-13-9360 kernel: NX (Execute Disable) protection: `active``

mastercaution, 2020-03-31:
Oh yes, you are totally right.
If you unplug your power supply and plug it back in when your battery is still fully charged, then the UPower state is `FULLY_CHARGED`. The extension updates the profiles only when the state is `CHARGING` and `DISCHARGING`. In case of the state being `FULLY_CHARGED` it only prints a message to the log, as we can see here:
`mars 31 05:41:09 dell-xps-13-9360 gnome-shell[2708]: Power state changed: fully charged` 

So it should theoretically work when you plug your power supply back in after the battery lost at least 1 percent, you can try it out :) .

This is a bug in the extension, I will fix it. I didn't notice that in my tests because my hardware stops charging the battery as soon as it's at 80% ... so apparently it never reached `FULLY_CHARGED`, but good that you noticed it ^^

fin-ger, 2020-04-08:
Damn it github... Reopen, as release is not out yet...

PR #111 (2020-04-04): Fixed fully charged bug #108
These little changes should fix #108 

@Tikilou would you test it for me?
closed 2020-04-14 by fin-ger