deinstapel/cpupower #112: gnome-shell-extension-prefs not working on Gnome 3.36.1

mastercaution, 2020-04-04:
`gnome-shell-extension-prefs` seems to be gradually replaced by `gnome-extensions prefs` ([Source](https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/2562)). 
In fact, `gnome-shell-extension-prefs` cannot open extension preferences anymore (in 3.36.1).
```sh
$ gnome-shell-extension-prefs cpupower@mko-sl.de

(gnome-shell-extension-prefs:52584): GLib-GIO-CRITICAL **: 17:04:52.343: This application can not open files.
```
I am currently on Arch Linux with Gnome 3.36.1 (where the issue occurs).
Ubuntu 20.04 is currently on 3.36.0 where `gnome-shell-extension-prefs` is still working fine. Despite the feature freeze of Ubuntu 20.04, it could be updated.

fin-ger, 2020-04-04:
Ah great! Will add this to #107 then, as I already added some version compat changes in there

mastercaution, 2020-04-04:
Keep the backward compatibility in mind. `gnome-extensions` is not available in Ubuntu 18.04 :wink:

commit 6167bea5 (2020-04-05): Fix #112
closed 2020-04-05 by fin-ger