deinstapel/cpupower #116: Profiles in the preference is empty

foldfelis, 2020-04-14:
Profiles in the preference is empty, even if I add one manually. 

Environment:
    OS: EndeavourOS Linux x86_64
    Kernel: 5.6.3-arch1-1
    DE: GNOME 3.36.1

[image]

fin-ger, 2020-04-14:
Hmm, it looks like some initialization code is broken. Can you please log out and back in again and upload the output of the following command:

```
journalctl /usr/bin/gnome-shell -n 300
```

> Please make sure to not include any sensitive information in the output (passwords, names, etc.)

foldfelis, 2020-04-14:
Hi @fin-ger logs as following:

```
-- Reboot --
 4月 14 18:50:42 msi gnome-shell[884]: Unset XDG_SESSION_ID, getCurrentSessionProxy() called outside a user session. Asking logind directly.
 4月 14 18:50:42 msi gnome-shell[884]: Will monitor session c1
 4月 14 18:50:43 msi gnome-shell[884]: Error looking up permission: GDBus.Error:org.freedesktop.portal.Error.NotFound: No entry for geolocation
 4月 14 18:50:43 msi gnome-shell[884]: JS WARNING: [resource:///org/gnome/shell/ui/layout.js 24]: reference to undefined property "MetaWindowX11"
 4月 14 18:50:43 msi gnome-shell[884]: Registering session with GDM
 4月 14 18:50:57 msi gnome-shell[1222]: Window manager warning: Overwriting existing binding of keysym ff1b with keysym ff1b (keycode 9).
 4月 14 18:50:58 msi gnome-shell[1222]: Unset XDG_SESSION_ID, getCurrentSessionProxy() called outside a user session. Asking logind directly.
 4月 14 18:50:58 msi gnome-shell[1222]: Will monitor session 2
 4月 14 18:50:58 msi gnome-shell[1222]: Telepathy is not available, chat integration will be disabled.
 4月 14 18:50:58 msi gnome-shell[1222]: loading user theme: /usr/share/themes/Arc-Dark/gnome-shell/gnome-shell.css
 4月 14 18:50:58 msi gnome-shell[1222]: Usage of object.actor is deprecated for Caffeine
                                         get@resource:///org/gnome/shell/ui/environment.js:285:29
                                         _init@/home/n/.local/share/gnome-shell/extensions/caffeine@patapon.info/extension.js:96:9
                                         enable@/home/n/.local/share/gnome-shell/extensions/caffeine@patapon.info/extension.js:316:25
                                         _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:165:32
                                         loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:316:26
                                         _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:549:18
                                         collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
                                         _loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:528:19
                                         _enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:558:18
                                         _sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:589:18
                                         init@resource:///org/gnome/shell/ui/extensionSystem.js:54:14
                                         _initializeUI@resource:///org/gnome/shell/ui/main.js:252:22
                                         start@resource:///org/gnome/shell/ui/main.js:141:5
                                         @<main>:1:47
 4月 14 18:50:58 msi gnome-shell[1222]: Usage of object.actor is deprecated for Caffeine
                                         get@resource:///org/gnome/shell/ui/environment.js:285:29
                                         _init@/home/n/.local/share/gnome-shell/extensions/caffeine@patapon.info/extension.js:145:9
                                         enable@/home/n/.local/share/gnome-shell/extensions/caffeine@patapon.info/extension.js:316:25
                                         _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:165:32
                                         loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:316:26
                                         _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:549:18
                                         collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
                                         _loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:528:19
                                         _enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:558:18
                                         _sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:589:18
                                         init@resource:///org/gnome/shell/ui/extensionSystem.js:54:14
                                         _initializeUI@resource:///org/gnome/shell/ui/main.js:252:22
                                         start@resource:///org/gnome/shell/ui/main.js:141:5
                                         @<main>:1:47
 4月 14 18:50:58 msi gnome-shell[1222]: Usage of object.actor is deprecated for Caffeine
                                         get@resource:///org/gnome/shell/ui/environment.js:285:29
                                         _init@/home/n/.local/share/gnome-shell/extensions/caffeine@patapon.info/extension.js:146:9
                                         enable@/home/n/.local/share/gnome-shell/extensions/caffeine@patapon.info/extension.js:316:25
                                         _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:165:32
                                         loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:316:26
                                         _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:549:18
                                         collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
                                         _loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:528:19
[... cut ...]
 4月 14 18:51:06 msi gnome-shell[1222]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
 4月 14 18:51:06 msi gnome-shell[1222]: Window manager warning: Overwriting existing binding of keysym ff1b with keysym ff1b (keycode 9).
 4月 14 18:51:06 msi gnome-shell[1222]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
 4月 14 18:51:06 msi gnome-shell[1222]: Window manager warning: Overwriting existing binding of keysym 35 with keysym 35 (keycode e).
```

fin-ger, 2020-04-14:
:see_no_evil: sorry, I forgot to mention to also open up the preferences dialog after login

foldfelis, 2020-04-14:
@fin-ger I just update the logs [above](https://github.com/martin31821/cpupower/issues/116#issuecomment-613375839)

fin-ger, 2020-04-14:
Hmm, still nothing suspicious there... Maybe the error is not logged to the gnome-shell... Let's try without filtering:

```
journalctl -n 500
```

Hopefully, the error is included in the output this time

Another thing you can try is open up [dconf](https://wiki.gnome.org/Projects/dconf) and check the contents of `/org/gnome/shell/extensions/cpupower/profiles`

[image]

The default content is

```
[(0,10,false,"Energy Saver", ""), (0,30,false,"Quiet", ""), (20,80,true,"Multimedia", ""), (50,100,true,"High Performance", "")]
```

foldfelis, 2020-04-14:
> Hmm, still nothing suspicious there... Maybe the error is not logged to the gnome-shell... Let's try without filtering:
>
> ```
> journalctl -n 500
> ```
>
> Hopefully, the error is included in the output [...]

This is odd, I can't find the field `profiles`
[image]

foldfelis, 2020-04-14:
> Hmm, still nothing suspicious there... Maybe the error is not logged to the gnome-shell... Let's try without filtering:
>
> ```
> journalctl -n 500
> ```
>
> Hopefully, the error is included in the output [...]

Also, I fond this:

```
 4月 14 20:30:37 msi systemd[1111]: Started Application launched by gnome-shell.
 4月 14 20:30:37 msi dbus-daemon[1140]: [session uid=1000 pid=1140] Activating service name='org.gnome.Shell.Extensions' requested by ':1.70' (uid=1000 pid=1785 comm="gnome-extensions prefs cpupower@mko-sl.de ")
 4月 14 20:30:36 msi PackageKit[960]: search-file transaction /10_ebdbdcdb from uid 1000 finished with success after 4ms
 4月 14 20:30:36 msi PackageKit[960]: search-file transaction /11_aeabebee from uid 1000 finished with success after 5ms
 4月 14 20:30:36 msi PackageKit[960]: search-file transaction /12_dbcdcdea from uid 1000 finished with success after 4ms
 4月 14 20:30:36 msi PackageKit[960]: get-details transaction /13_beabdcba from uid 1000 finished with success after 2ms
 4月 14 20:30:37 msi systemd[1111]: Started Application launched by gnome-shell.
 4月 14 20:30:37 msi dbus-daemon[1140]: [session uid=1000 pid=1140] Activating service name='org.gnome.Shell.Extensions' requested by ':1.70' (uid=1000 pid=1785 comm="gnome-extensions prefs cpupower@mko-sl.de ")
 4月 14 20:30:37 msi dbus-daemon[1140]: [session uid=1000 pid=1140] Successfully activated service 'org.gnome.Shell.Extensions'
 4月 14 20:30:37 msi gjs[1789]: JS ERROR: ReferenceError: global is not defined
                                 status@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:80:9
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:135:18
                                 _init/createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529:46
                                 onShowArrowSwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:403:24
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:94:30
                                 _init/createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529:46
                                 onShowIconSwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:398:24
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:91:29
                                 _init/createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529:46
                                 onUseGHzInsteadOfMHzSwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:393:24
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:88:39
                                 _init/createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529:46
                                 onShowCurrentFrequencySwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:387:24
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:85:41
                                 onMainWidgetRealize@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:378:14
                                 OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:139:20
                                 asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
                                 run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
                                 main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
                                 run@resource:///org/gnome/gjs/modules/script/package.js:222:19
                                 start@resource:///org/gnome/gjs/modules/script/package.js:206:5
                                 @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
 4月 14 20:30:37 msi gjs[1789]: JS ERROR: ReferenceError: global is not defined
                                 status@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:80:9
                                 onShowArrowSwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:404:14
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:94:30
                                 _init/createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529:46
                                 onShowIconSwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:398:24
                                 _updateSettings@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:91:29
                                 _init/createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529:46
                                 onUseGHzInsteadOfMHzSwitchActiveNotify@/home/n/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:393:24
[... cut ...]
                                 asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
                                 run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
                                 main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
                                 run@resource:///org/gnome/gjs/modules/script/package.js:222:19
                                 start@resource:///org/gnome/gjs/modules/script/package.js:206:5
                                 @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

```

fin-ger, 2020-04-14:
Umm, does anyone here know why `global` is undefined?

/cc @martin31821 @mastercaution

milos7250, 2020-04-15:
I second this issue, I can confirm that I have the same problem, where I can create the profiles in the settings and set default profiles for AC/Battery, but after closing the settings they all go away.

I see the same error "global is not defined" as @foldfelis does. I am able to set profiles through dconf and use them from the extension, I have not been able yet to set them to change automatically, as I have no idea where to find the profile IDs.

fin-ger, 2020-04-15:
@milos7250 Can you please provide me with your system details (Distro, Kernel, Gnome version, Wayland/X version). The error occurs because the system preferences dialog is logging information to the system logger over the `global` object in gjs. However, it seems to be missing on your system, which is something we cannot reproduce (Fedora 32, Ubuntu 18.04/20.04, Gentoo Linux Gnome 3.34, Arch Linux Gnome 3.36.2).

milos7250, 2020-04-15:
System:
  Host: --- Kernel: 5.6.3-2-MANJARO x86_64 bits: 64 
  Desktop: Gnome 3.36.1 Distro: Manjaro Linux 
DM: X11 X.Org version: 1.20.8

fin-ger, 2020-04-15:
I actually don't know why we used the `global` object. It seemed to be the way to log to the system logger when we started this extension. I replaced it with the logging method mentioned in the [docs](https://wiki.gnome.org/Projects/GnomeShell/Extensions). I found no reference on why the `global` object is present on some systems and absent on others.

I will start testing several systems with these changes and check if everything works.

@foldfelis @milos7250 Can you check whether this fixes the extension on your system? Use the [developer installation method (option 1)](https://github.com/martin31821/cpupower#option-1) to test. The gnome shell can be restarted with `Alt+F2` command: `r` on X based installations. On Wayland you need to log out and back in.

milos7250, 2020-04-15:
Yes, this fixed the issue. I can see the default profiles in settings, modify them and set them to ac/battery as well. Auto switch works as well. There are no errors in the journalctl.

Thank you very much for fast fixing :). Seems that you can close this now.

fin-ger, 2020-04-15:
Nah, still a lot of testing to do on other system. May take a week or so for releasing the fix

foldfelis, 2020-04-15:
@fin-ger The profile are now works like a charm

fin-ger, 2020-04-15:
Need this for tracking :wink:

fin-ger, 2020-04-15:
I verified for the following setups:

 - Fedora 31
 - Ubuntu 18.04
 - Ubuntu 19.10
 - Ubuntu 20.04 (with gnome-shell-extensions installed)
 - Gentoo Linux with Gnome 3.34

I failed to verify Debian 10 with Gnome 3.34 as the network is broken on that machine.

Also, I need someone to test it on latest Arch, my installation is missing a working graphics driver :sweat_smile:

mastercaution, 2020-04-15:
I can confirm this issue. 
It appears an Arch Linux with Gnome 3.36.1 and Ubuntu 20.04 also with Gnome 3.36.1 but I could not reproduce it on Ubuntu 20.04 with Gnome 3.36.0.

> Also, I need someone to test it on latest Arch

Your fix works on Arch Linux (5.6.4-arch1-1) with Gnome Shell version 3.36.1

fin-ger, 2020-04-15:
Ok, so let's release a patch

avanishsubbiah, 2020-04-17:
I have the same issue on Arch x86_64 on the 5.6.4 kernel with Gnome Shell version 3.36.1.

fin-ger, 2020-04-17:
The new version is on its way. Please be patient

fin-ger, 2020-04-21:
Update got reviewed and released

closed 2020-04-21 by fin-ger