deinstapel/cpupower #117: Add CPU temperature support and dynamic profiles

NedkoHristov, 2020-04-21:
Great plugin, thanks @martin31821 ! I'm using it on a daily basis on Lenovo X1 yoga and it's must-have.

Do you have plans to add CPU temp? If so there can be two advantages:
* Show CPU temp (no kidding...);
* Dynamic profiles - As user preference - if temp goes up then lower the max frequency or adjust on
     * If temp data is not present then just disable/don't show the option).

fin-ger, 2020-04-21:
Hey @NedkoHristov, we're glad we can support you in your daily work!

Regarding CPU temperature display, we won't support this as we are solely focused on displaying and controlling the CPU frequency (currently only intel supported, this should change in the future). Please refer to other extensions like [Freon](https://extensions.gnome.org/extension/841/freon/) for CPU temperature display. It's the old Unix philosophy :grin: One tool for one job.

Dynamic profiles are on our todo list, please see #62. However, this feature currently has no high priority, so it may take a while to land in `cpupower`. This would of course need cpupower to handle CPU temperatures, but only for controlling the frequency. Temperature display is not planned.

NedkoHristov, 2020-04-21:
Thanks for the detailed feedback, @fin-ger. Much appreciated :bow:

closed 2020-04-21 by NedkoHristov