deinstapel/cpupower #118: after update from 18.04 to 20.04 the extension is still there but no showing in the TOP menu

petr-hybler, 2020-04-22:
Hello, I have just recently updated to 20.04 from 18.04 and the extension is still there but I cannot seem to get it to the TOP bar

[image]
[image]

I can't even seem to find it in the software center 

[image]

OS Model: Ubuntu 20.04 LTS (x64)
CPU Model: Intel® Core™ i7-8565U CPU @ 1.80GHz × 8 
Gnome: 3.36.1
Kernel: Linux version 5.4.0-26-generic (buildd@lcy01-amd64-029) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2))
policyKit: pkaction version 0.105

fin-ger, 2020-04-22:
Can you please run

```
journalctl -f
```

and toggle the extension off and on again?

Please paste the output of the above command here. Make sure, no confidential information is included in the output, like passwords, names, location, etc.

Regarding the software center, you can try to reinstall the extension over the [extensions website](https://extensions.gnome.org/extension/945/cpu-power-manager/).

fin-ger, 2020-04-22:
Oh, and I forgot to mention. For Ubuntu 20.04 it __might__ be necessary to install the [`gnome-shell-extensions`](https://packages.ubuntu.com/focal/gnome-shell-extensions) package from the [`universe`](https://help.ubuntu.com/community/Repositories/Ubuntu) repository.

petr-hybler, 2020-04-22:
I already have the gnome-shell-extensions ... 

the output of this command is as follows: 

`Apr 22 16:15:39 lcf gnome-shell[1892]: JS ERROR: Error: Tried to construct an object without a GType; are you using GObject.registerClass() when inheriting from a GObject type?
                                       CPUFreqProfileButton@/home/petr/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profilebutton.js:37:9
                                       createMenu@/home/petr/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:134:33
                                       CPUFreqIndicator@/home/petr/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:83:14
                                       enable/</<@/home/petr/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:73:34
                                       spawn_process_check_exit_code/<@/home/petr/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:50:13
Apr 22 16:15:39 lcf gnome-shell[1892]: Could not create transient scope for PID 47141: GDBus.Error:org.freedesktop.DBus.Error.UnixProcessIdUnknown: Process with ID 47141 does not exist.
`
I only inserted the error that occurred .. if you want the whole log, let me know .. I can paste it

fin-ger, 2020-04-22:
This looks like the issue fixed in be89b51. Are you able to update/reinstall the extension via the website?

petr-hybler, 2020-04-22:
sorry am a bit struggling here .. I downloaded 3.36/20 ... but dunno how to "update" ?

fin-ger, 2020-04-22:
[image]

The extensions website should include a green update button (I'm on the latest version, so no update available). It should not be necessary to download the update manually.

petr-hybler, 2020-04-22:
not for me either :( 
[image]

plus when I tried "sudo make install install-tool" it returned this:

`Compiling schemas...
Updating translations...
./scripts/update-translations.sh: line 9: /usr/bin/msgfmt: No such file or directory
make: *** [Makefile:14: build] Error 127
`

fin-ger, 2020-04-22:
It is not necessary to build the extension yourself. As you are currently running the extension as a user installation, I would recommend not installing it system wide now. The extension website says you need to install a browser extension in order to manage your extensions here. Please do so. The website will then be able to manage your extensions and update accordingly. I can't tell why the Software Center is not capable of this.

petr-hybler, 2020-04-22:
I need browser extension to manage it in the system??

fin-ger, 2020-04-22:
Yes, or you can use the `Gnome Web` Browser, which has this functionality included.

petr-hybler, 2020-04-22:
still nothing in the top panel ... how can I uninstall this? I will try to remove and install again

petr-hybler, 2020-04-22:
I even managed to uninstall via the website and extension .. and install again .. but still nothing .. same error in journalctl ...

any idea?

fin-ger, 2020-04-22:
To clarify: Using the extensions website for updating/installing/managing gnome-shell extensions is the default for all gnome extensions. The Software Center can also be used for extension updates/management, but as gnome-shell-extensions are not fully supported under 20.04 yet, this is not working.

The full uninstallation got implemented in version 9.0.0, you seem to have a pre 9.0.0 version. After you updated the extension on the website with the green button, you should be able to migrate your installation to the v9 version.

If this is not working, you can try to forcefully remove it by running

```
rm -r ~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de
```

Now log out and back in again.

petr-hybler, 2020-04-22:
fantastic :) it's fixed !!!

[image]

fin-ger, 2020-04-22:
I'm sorry that the Ubuntu Software Center is not working for extension management. You may report this to the Ubuntu devs, as I cannot do anything about it but advise you to avoid the Software Center for extension management :(

petr-hybler, 2020-04-22:
nvm.. I know it's not your fault ... thx a lot for your time and help :) I am happy to have this awesome extension :) thank you !!!

closed 2020-04-22 by fin-ger