deinstapel/cpupower #119: Extension not loading on Fedora 30 (Gnome  3.32.2)

dalessan, 2020-04-30:
Hi,

I love this extension, but it's not working anymore on my system (Fedora 30, Gnome 3.32.2).

I tried re-installing it using the instruction in the readme, but it didn't work.

Here's the error I can see from journalctl:
```
Extension "cpupower@mko-sl.de" had error: TypeError: GObject.registerClass() used with invalid base class (is PopupMenuItem)
```

Regards,
Daniele

martin31821, 2020-04-30:
Did you get the latest update? Maybe it breaks some older gnome versions.

fin-ger, 2020-04-30:
It could be that my information I found was wrong regarding the integration of GObject in gnome-shell. I assumed it got integrated in 3.32, but apparently it's not available... Sorry for that!

Can you install the latest version and open up `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profilebutton.js` and change `3.30` to `3.32` in [line 45](https://github.com/martin31821/cpupower/blob/master/src/profilebutton.js#L45)? Log out and back in again and try to enable the extension again.

Hope this helps!

dalessan, 2020-04-30:
@fin-ger  changing that line allowed the extension to load, but after installing the policykit action it broke again. The error on journactl was different though (sorry I didn't think of saving it!).

@martin31821  switching back to what is reported a v18 on https://extensions.gnome.org/ solved the problem.

fin-ger, 2020-04-30:
@dalessan as v18 includes a security issue, I would ask you to check the error which occurred in journalctl, so we could release a fix for gnome 3.32.

dalessan, 2020-04-30:
@fin-ger sure, I'll do that. Might have to wait until the weekend though. Hope that's fine.

dalessan, 2020-05-02:
@fin-ger, here is the error I see in the log:
```
JS ERROR: Error: Wrong type Object; string expected
PopupMenuItem<@resource:///org/gnome/shell/ui/popupMenu.js:245:22
CPUFreqProfileButton@/home/dalessan/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/profilebutton.js:830:19
createMenu@/home/dalessan/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:167:33
CPUFreqIndicator@/home/dalessan/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:9
enable/</<@/home/dalessan/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:96:34
spawn_process_check_exit_code/<@/home/dalessan/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:63:13
```

dalessan, 2020-05-03:
Hi, I think I found what the issue was and created a pull request: #120

Note, however, that I couldn't test it on Gnome > 3.32, so you may want to check that my fix doesn't break newer versions of Gnome.

Also, it's the first time I touch JS code, so my fix might not respect best practices or coding style for JS. Sorry about that.

fin-ger, 2020-05-03:
Your changes are fine, I merged them. Works well on my Gentoo system. I will wait for someone to test it on latest Arch Linux and Ubuntu 20.04 and will then release a patch version.

@jwuensche @martin31821 Arch test?

jwuensche, 2020-05-03:
@fin-ger Works for me, with 3.36.2

closed 2020-05-11 by fin-ger