deinstapel/cpupower #123: Turning off turbo mode intel i7950h

Apogeum12, 2020-05-20:
Ubuntu 20.04 on 19.10 this same
kernel 5.4.0-31-generic on 5.4.0-26 this same
Installed in system tlp, and intel-undervolt
Lenovo Legion y540

If I return from Battery mode to quiet mode or performane mode etc.(AC power ofcourse)  then turbo mode the turn off(in setting as enabled), Even when I try to enabled it manualy.
When I run for example fish benchmark(WebGL Aquarium) in browser.   Turbo mode is off, he not change automatic for a long time. After 30-60 secunds  turbo  mode something it turn on but return to off  if i exit the benchmark.  So my notebook work on (1.5-2.6 Ghz)  in  quiet mode but should (1.5-2.8  Ghz)  2.6  GHz is base frequencies.

Can anybody help me,or this is bug the cpu power managment  ?

fin-ger, 2020-05-27:
Have you configured Auto-Switch in CPU Power Manager?

[image]

Your Auto-Switch profiles may have Turbo Boost configured, and will change your Turbo Boost setting when switching from e.g. battery to AC. You can turn Auto-Switch off in the CPU Power Manager menu to prevent profile switching.

> When I run for example fish benchmark(WebGL Aquarium) in browser. Turbo mode is off, he not change automatic for a long time. After 30-60 secunds turbo mode something it turn on but return to off if i exit the benchmark.

This behavior is not created by CPU Power Manager. This extension will only reset a manually set Turbo Boost setting directly after switching from AC to battery or vice versa. After that you can re-enable or disable Turbo Boost in the extension menu. If this is not working, it's an issue outside of this extension.

Apogeum12, 2020-05-27:
You have right, thank for your answerr. When I changed in setting tlp
PCIE_ASPM_ON_AC= powersave  --> performance 
and
CPU_SCALING_GOVERNOR_ON_AC= powersave -> performance
and without changing
CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
In add I deleted from system package "irqbalance"
but I don't sure done this help some in this issue.
Now turbo mode enable more faster.

closed 2020-05-28 by fin-ger