deinstapel/cpupower #124: Query related to Vulkan and CPU Power Manager

deepdivingspearo, 2020-05-26:
# Creating a new issue

* Check if another issue with the same problem exists
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
  * policykit working and rule working
* Check if the cpufreqctl utility works correctly
* Please include the following information (only for bugreports):
  * CPU Model
  * Linux kernel version
  * policykit version
  * Gnome shell version
  * journalctl -xb (maybe pastebin/filtered)

deepdivingspearo, 2020-05-26:
Great extension, thanks for the work on this. 
I recently learned that Ubuntu 2004 has the latest MESA drivers and a bunch of optimisations for Vulkan which kick in for gaming (if the game is supported). My question was, will the optimations for Ubuntu 2004 and Vulkan automatically override my default / current CPU setting in "CPU Power Manager"? I would want the OS to do this by default as I want max power for gaming, was just wondering if it was the case.

fin-ger, 2020-05-27:
I don't know if Mesa / Ubuntu changes the CPU frequency scaling for gaming. However, CPU Power Manager will display all changes made to the CPU frequency scaling. So, if Ubuntu or Mesa do change the CPU frequency, you can see it in CPU Power Manager.

deepdivingspearo, 2020-05-28:
Thanks for the info, much appreciated.

closed 2020-05-28 by deepdivingspearo