deinstapel/cpupower #125: Security warning prevents me from using it

ferkulat, 2020-06-08:
Hi, 
today I was faced with following warning:
"Warning: Security issues were found with your installation!"
"Please update immediately!"
And the only action available:
"Attempt tool update"

This was very unfortunate:
1. I was on battery and wanted to switch to power safe mode, but could not.
2. I was on mobile connection. I did not want to use my limited data volume for updating software.
3. The attempt to update, asked me for the root password, which I suspected as trick to get control over my system. So I did not enter it. Because the only way for me to update software, is via the package manager.

For me it would be totally fine to see this warning and update an hour later when I am back at home.
And for me the most important thing was: I wanted to use the tool at this very moment.

Please trust your users, and treat them like grown ups, that can make the decision "when to update" them self.
It would be totally ok to show the warning over and over when ever I click on the applet. But please let me use the tool.

With kind regards

fin-ger, 2020-06-10:
Thanks for your report!

This security warning should only be displayed if

1. previously installed via the extensions website
2. not properly installed by the package manager

If 1)
You can do the tool update without an internet connection. Root is necessary for updating the policykit rules as they included security issues. This is the same for a clean install from the extensions website.

If 2)
Please contact your distributions package maintainer and inform them of improper packaging of the policykit rules. Alternatively, you can give us a hint which package you installed so we can assist the package maintainer in properly packaging this extension.

Best regards,
Fin

ferkulat, 2020-06-14:
Thank you very much.
The package was not updated, but a user came up with a solution:
https://aur.archlinux.org/packages/gnome-shell-extension-cpupower-git/
Now I can use it again.

closed 2020-06-14 by ferkulat