deinstapel/cpupower #130: Turboboost Autoswitches On

fernrees, 2020-07-27:
My turbo boost keeps auto-switching to on.  I have turned of the auto switch option and changed all the profiles to turn auto switch off.  Despite this it keeps turning on the turbo boost.

mastercaution, 2020-07-28:
Can you give us a little more information about your system? Gnome version, distro, CPU, ...

fernrees, 2020-08-01:
Processor: Intel® Core™ i7-9750H CPU @ 2.60GHz × 12 
Computer XPS 15 
Distro 18.04.4 LTS
Gnome info:

<?xml version="1.0" encoding="UTF-8"?>
<gnome-version>
 <platform>3</platform>
 <minor>28</minor>
 <micro>2</micro>
 <distributor>Ubuntu</distributor>

</gnome-version>

mastercaution, 2020-08-01:
Okay, lets track down the issue by firstly checking if the kernel can turn off Turbo Boost. To do that, try to turn off the turbo manually:
`sudo sh -c "echo '1' > /sys/devices/system/cpu/intel_pstate/no_turbo"`

Turbo Boost should now be turned off. Verify it with the extensions UI or:
`cat /sys/devices/system/cpu/intel_pstate/no_turbo` (output should be 1)

If this works you can turn Turbo Boost back on with:
`sudo sh -c "echo '0' > /sys/devices/system/cpu/intel_pstate/no_turbo"`
If not, try again after disabling the extension + restart.

If you are able to control Turbo Boost with these commands then the issue lies in the extension. I'm not really familiar with Ubuntu 18.04 and Gnome 3.28 but let's see what your results are first ;)

fernrees, 2020-08-02:
Yes able to turn it off and on with commands above.

fernrees, 2020-08-02:
Also very appreciative of the effort!

mastercaution, 2020-08-05:
Unfortunately, I was not able to reproduce the error. I tested it on Ubuntu 18.04.4 and Gnome Shell 3.28.4 (you probably have the same Gnome **Shell** version).

To better understand this issue I would like to ask you how did you install the extension? Did you install it via the Gnome Extensions website or did you run the install script manually (which arguments you used then?)?

What about the frequency sliders? Do they work correctly? If you don't know for sure you can check it just like the turbo state with:
`cat /sys/devices/system/cpu/intel_pstate/min_perf_pct` and
`cat /sys/devices/system/cpu/intel_pstate/max_perf_pct`
There you should read the same values you configured with the sliders in the popup menu (0-100)

fernrees, 2020-08-12:
Those also reset sometimes, but those commands work.

fernrees, 2020-12-05:
I ended up upgrading to Ubuntu 20 and don't experience any issues.  Runs perfectly.

fin-ger, 2020-12-08:
Alright, it's not nice that over time we get regressions into old distribution versions but I don't see how to effectively avoid this. If anyone else still runs into this issue, please don't hesitate to reopen!

closed 2020-12-08 by fin-ger